opentx/opentx #5543: OTX 2.2.1 R9M No chance to enable telemetry on TX

Allerhopp, 2017-12-19:
2.2.1, X9D, R9M LBT (EU)
With checkbox telemetry enabled in companion telemtry works. The radio itself has no chance to enable telemetry. It is offered in bind-options, but does not work, if one selects it.
You have to use companion first, to enable telemetry there, then it works on the radio. Why not enable  as default? You can always disable with bind option (i hope ;)).

3djc, 2017-12-19:
Telemetry on R9M only works if internal module is disabled,  have you checked it is ?

Allerhopp, 2017-12-19:
Sure, internal disabled, using LBT (EU) version. As i wrote, telemetry works. But only if the checkbox is enabled using Companion. Bind option in TX offers telemetry mode, but it does not work, without the Companion workaround. TX has no access to the checkbox, Companion has.

3djc, 2017-12-19:
OK, understood

3djc, 2017-12-19:
As a side note, you can workaround with radio only : select fcc module, activate module telem, choose EU again, the settings should still be 'active'

kilrah, 2017-12-19:
> Why not enable as default? You can always disable with bind option (i hope ;)).

Module telemetry and receiver telemetry are not the same thing

Allerhopp, 2017-12-19:
Leniency please dev´s, i was in a deep depression because of the R9 performance ;)

closed 2017-12-29 by bsongis

LeeHudsonDLS, 2018-08-13:
> 2.2.1, X9D, R9M LBT (EU)
With checkbox telemetry enabled in companion telemtry works. The radio itself has no chance to enable telemetry. It is offered in bind-options, but does not work, if one selects it.
You have to use companion first, to enable telemetry there, then it works on the radio. Why not enable as default? You can always disable with bind option (i hope ;)).

@Allerhopp Hi, i'm having no lucjk getting telemetry working with an R9M. Where is this checkbox in companion that you speak of? I can't see it anywhere.
Thanks
Lee

Allerhopp, 2018-08-13:
Are you using 2.2.2? It is working different now. SPort link is enabled, if internal module is disabled and vice versa.

[image]

PR #5553 (2017-12-29): R9M : activate automatically MODULE telem when possible
With this change, user doesn't have to deal with module telem anymore : if possible, module telem is enabled. Status of said telemetry is displayed as read only

@mpaperno, if accepted, could you do the Companion part ? I'm unsure if we just want to hide module telem altogether, or if we want to replate radio behaviour (read only status). The previously used setting bit is back to spare

This also fixes #5543
commit 8ee9199a (2017-12-29): R9M : activate automatically MODULE telem when possible (#5553)

* Fix #5543

* Auto activate R9M telem

* Adjust radio ui

* Thx @schwabe

* Stop being stupid ?

* PR rework

* [Companion] Removed "Module Telemetry" checkbox.