opentx/opentx #5546: 2.2.1 Mix Scripts Custom Screen X7 Not showing all outputs

RCdiy, 2017-12-19:
2.2.1 Companion
Mix Scripts
Simulator
Custom Scripts
Scripts selected "Sqnc"
Problem
X7 Outputs shows only 1 of 2
X9D Outputs shows both
Anone reported this?
Not all outputs showing for mix scripts on X7
[image]

[image]

RCdiy, 2017-12-19:
Same behaviour on the tx

kilrah, 2017-12-19:
No scrolling down?

RCdiy, 2017-12-19:
Correct... Scrolloing down wraps back to the top...

RCdiy, 2017-12-19:
Sample file 
[Sqnc.lua.zip](https://github.com/opentx/opentx/files/1572445/Sqnc.lua.zip)

janbolt, 2018-01-05:
Confirm this.
Custom script screen shows always 1 output less than the script has.
For a script with 6 outputs, only 5 will be shown,

RCdiy, 2018-01-05:
Yes. In sample I believe 2 outputs exist. 1 is seen.

PR #5601 (2018-01-26): Fix custom scripts outputs handling
One output was missing if there where inputs lines
Scrolling was making a pause at Outputs label

This fixes #5546
closed 2018-01-26 by bsongis