opentx/opentx #5547: Taranis SD Card Screen Does Not See Firmware with Multimodule Build Option

helinv, 2017-12-19:
Opentx 2.2.1 Released Version
Taranis X9D+
Build Options: Lua, Multimodule

The SD Card screen (after Radio Setup screen) does not show firmware files with Multimodule build option.

3djc, 2017-12-19:
Name of the firmware is too long, needs to be renamed on the PC/Mac

closed 2017-12-19 by 3djc

helinv, 2017-12-19:
I had already tried that.  It does not help.  If firmware Build option contains Multimodule the radio does not see the file.

helinv, 2017-12-19:
As an example, the radio sees this file
opentx-x9d+-multimodule-lua-en-2.2.0.bin
,but not this file with the exact same character count.
opentx-x9d+-multimodule-lua-en-2.2.1.bin

Any combination of multimodule build option causes the file to not be seen in the radio regardless of filename length.

3djc, 2017-12-19:
In that screen the content of the file is not analyzed at all, just names are displayed. There is something wrong somewhere , but hard to tell  exactly what. Could you upload your sd folder content ?

3djc, 2017-12-19:
Yes it is size, remove 2 or 3 chars and it works

helinv, 2017-12-19:
Some issues never cease to amaze me.  My radio would see the file 
opentx-x9d+-multimodule-lua-en-2.2.0.bin with no problem.  I could even select it to flash my bootloader.

I deleted all files under FIRMWARE, and added them back.  Now my radio sees the shorter filenames, even with multimodule, but now doesn't see opentx-x9d+-multimodule-lua-en-2.2.0.bin.

Issue fixed but not solved (SD card problem).

Thanks for your time.  It is most definitely appreciated.

3djc, 2017-12-19:
I think you saw one with a similar but shorter name. You cannot see that file name (you can try with simu as well)

helinv, 2017-12-19:
I actually copied the name right from the SD card.  Who knows?

Thanks again.