opentx/opentx #5558: Companion N466-467 radio.bin issue with X7

capitaineflam25, 2017-12-23:
You maybe aware of that but reading model and settings from radio (X7) from Companion N466/N467 result in an error.

[image]

3djc, 2017-12-23:
indeed

r41065, 2017-12-30:
Same Problem with Taranis X9d and Open Tx 222N473, not testet with Versions below.
Ralf

3djc, 2018-01-05:
this my breaking, the fix misses also an issue somewhere else, making one

capitaineflam25, 2018-01-07:
I'm still facing the issue with Companion N482 (2017-01-07), should i upgrade the radio firmware too (Actual eeprom version is 218 on my X7)

[image]

capitaineflam25, 2018-01-18:
Can you reopen the issue or should i create a new one ?

bsongis, 2018-01-18:
@3djc do you know which commit broke this?

3djc, 2018-01-18:
yes and no. It was the PCBFLAMENCO that caused it, I failed. So I fix the failuere, but it seems to have creeped back, but a number of changes to those two files (boards..h board.cpp) have happened as well

bsongis, 2018-01-18:
ok I will have a look, but during this week-end, I won't have enough time tonight

capitaineflam25, 2018-01-20:
Thanks a lot, this time it's OK (Tested on latest NB : N494)

3djc, 2018-01-20:
Yeah, we had two different, completely unrelated issue creating the same sympthom, that did make the process more complexx, sorry

commit e50fc202 (2018-01-05): Fix #5558