opentx/opentx #5560: LUA drawRectangle() crashes Simulator/Firmware when x is offscreen

capitaineflam25, 2017-12-23:
Simulator or radio firmware is crashing when using LUA function **drawRectangle()** out of screen on X7.

To reproduce the issue, open "crash.otx" (Find it here : [crash.zip](https://github.com/opentx/opentx/files/1584204/crash.zip)) on Companion, simulate radio and press "Menu" long.
Telemetry script "crash.lua" should load and simulator should crash.

The crash is created by drawing a rectangle at x=140 (Screen width for the X7 is 128)
Tested on N463

[image]

3djc, 2017-12-23:
While we do try our best to protect the radio against off screen drawing while maintaining decent performance, please be aware that there are loopholes and you can crash a radio with off screen drawing. You are warned.

capitaineflam25, 2017-12-23:
Off course, know i realised the point, i've added controls on my code to avoid writing offscreen but didn't know if i should report you that.

3djc, 2017-12-23:
Yes you should, I was just raising awarness on this risk

closed 2019-11-15 by stale[bot]