opentx/opentx #5561: X10S opentx v2.2.1 very loud noise at speaker, but now noise with FrOS

hrenz, 2017-12-23:
Its at a serie of the latest new X10S with openTx V2.2.1 
very very loud noise  at speaker with opentX V2.2.1
But no noise to hear with FrOS

Mybe FrOS swith the amplifer to Mute (off) if not inuse

could anyone test it

3djc, 2017-12-23:
Are you comparing with the same modules transmitting at the same power ?

hrenz, 2017-12-23:
yes, all same, 3 new X10S with this effect 
Fa. Engel has same problem with X10s and opentx
Muting the X10S  with FrOS is only speculation of me

ChristiaanVis, 2017-12-25:
I can confirm this issue. No remedy found so far; I tried the following:
- turned off RF
- switching internal or external antenna
- rerouted bluetooth antenna cable
- replaced hello and welcome chimes
- reduced background image size to 4kb
- flashed using both windows 10 and macOS using latest firmware (17 dec 2017)

Possibly issue relates to wav (OpenTx) versus mp3 (fros)?

Available to any further ideas to find the issue.

raphaelcoeffic, 2017-12-26:
Turning the internal module off does it for me. I guess the speaker wires are not shielded, as usual....

1Sven, 2017-12-26:
I think that's not new. The built-in mp3 player was killed with opentx. The audio output is affected. - It certainly can not change without serious hardware change. May be, this is perhaps the price for the simplified audio treatment. You should not be bothered by it, or have to take fros for certain applications.

schwabe, 2017-12-27:
@1Sven the mp3 player was not killed. It was never implemented.

kilrah, 2017-12-27:
It's possible it could be improved by muting the audio amp when nothing is playing.

raphaelcoeffic, 2017-12-29:
@kilrah I'd be glad to test it ;-)

3djc, 2017-12-29:
Didn’t @projectkk2glider try something like this already ?

RoninMedia12, 2018-01-14:
I can Confirm my x10s does this same thing my x12s and Taranis x9d plus do not all on 2.2.1,  But my Taranis did it before  I remember a while back I think that was on 2.1 or previous though  then one update just went away and has not returned on the x9d plus

voyn238, 2018-02-26:
Same issue here on new X10S. I put a sticker over the speaker which reduces the hiss to near inaudibility and reduces volume of audio cues, but it's still there. I didn't hear this hiss during the brief time the radio still had FrOS installed. It's most annoying at home when background noise is low, but you can still hear it in the field, even with the sticker.

bajamo, 2018-03-04:
Dear all, I have the same issue after put open tx on FRSKY HOrus x10s. What is the fix for this issue ?

3djc, 2018-03-04:
I'm afraid at this point in time there is none

bajamo, 2018-03-04:
on opentx 2.2 it is the same issue ?
thanks for your help

sw3Dp, 2018-03-15:
this is a bigger issue than just rf noise in the speaker as my day old x10s flashed to 2.2.1 has the speaker noise, rf crazy interference with pc, cell phone, all devices in my pc room.  to top it off it corrupts hal gimbal output making my radio completely non-functional, (that isn't just rf noise.)  FrOS is fine but damb it took me 4 years just to learn the basics of opentx, and i was really happy with the x9d ... mostly.

3djc, 2018-03-15:
That seems like a completely diffrent issue altogether, and not one that we have heard off

sw3Dp, 2018-03-15:
3djc seems different but i remember in 2015 my taranis with one of the intermediate builds had the exact same "noise" glitches that were quickly updated by the crew.  not a stretch to assume corrupted module function affecting hal, but that's just Dirk Gently speaking.

svsc99, 2018-04-04:
I get the same problem with my Horus x12s, OpenTX 2.2.1 . The error (noise) appears only with the external module **TBS Crossfire micro TX** version 1.25. With the orange module for DSM2 doesn't appear the problem. Are there any new fixes for the problem?

PR #5875 (2018-05-08): [X10] mute audio when fifo buffer is empty
Fixes #5561 by muting the X10 audio amplifier when no sound is being played. On targets with SOFTWARE_VOLUME, no audio is pushed to the queue when volume is muted.

This has been tested on X10 only.
closed 2018-05-08 by bsongis