opentx/opentx #5565: OPENTX 2.2.1 R-XSR "NO DATA"

HotwiresFPV, 2017-12-24:
I just flashed OpentTX 2.2.1 and I'm not getting any telemetry data going through. however if I revert to OpenTX 2.2.0 I get telemetry data.

schwabe, 2017-12-24:
We need more details though. What remote. No telemetry data at all or just some missing?

HotwiresFPV, 2017-12-24:
Taranis QX7. No sensor will populate data.

HotwiresFPV, 2017-12-24:
Taranis QX7 and r-xsr using the uninverted smartport pads on top of the
receiver. No sensor will populate data. Lua script does work.

schwabe, 2017-12-25:
Doyou get rssi from the rx?

trojanGoat, 2017-12-31:
Do you need to possibly update the firmware?  I just installed an r-xsr on my babyhawk and I got telemetry working.
http://rotorgeeks.com/index.php?route=product/product&product_id=877

HotwiresFPV, 2017-12-31:
Let me try that. Thanks

HotwiresFPV, 2018-01-27:
I got telemetry working now but I keep getting "telemetry lost" and "telemetry recovered" sometimes "sensor lost" warning. these are happening to a 4 drone with rxsr with latest firmware with PID support.

rollysing, 2018-03-03:
@HotwiresFPV Your Tx is swamping the Rx. Move it a little farther away and it would stop the constant telemetry drops.

Good luck.

djdazzy, 2018-03-08:
I have had the same issue on my x9d with a d6fr since upgrading to 2.2.1. Intermittent telemetry losing everything including RSSI and A2. I have flashed the internal XJT module with various INT(FCC) FW but it is still the same. I can bind the receiver to my X7 (on 2.2.1) or Devo 7E with no issue.

closed 2018-03-31 by HotwiresFPV

goodnerdboy, 2018-04-18:
Hi there.
HotwiresFPV did you solve your issue?
I have a r-xsr and qx7 doing the telemetry lost/telemetry recovered loop and can’t find a way to solve it...
I am using uninverted s.port. Tried both with a SBUS/S.port setup and with F.port, but both have the same result...

BombowaKuna, 2018-05-21:
I have the same problem. I'm getting telemetry and signal lost sometimes when i fly max 100 meters away. Got the 2.2.1 Open TX update for my Q X7S and 2017-11-09 171103 for R-XSR (all EU ver.).

Any ideas what's going on?

kilrah, 2018-05-21:
This doesn't seem like the same problem at all, but rather faulty hardware.

BombowaKuna, 2018-05-21:
so, you telemetry lost does not combine with signal/control lost? If not that im sorry, i misunderstood

kilrah, 2018-05-21:
No. 

IF you get a control loss then you also get a telemetry loss. Not the opposite.