opentx/opentx #5577: Horus X10 TBS Crossfire Micro TX Module

jcutcher01, 2017-12-29:
I recently noticed two issues with the TBS Crossfire TX Module (100mW micro) and the micro receiver. I am running a Horus X10 with 2.2.1. 

1) When I power up the radio for the first time and then execute the CROSSFIRE LUA script the window shows "Waiting for Devices" and will not disappear until I move either the sticks or switches several times. Then it seems to "wake-up" the TX module and the Crossfire TX menu item appears. With the radio turned on at this point, going in to and out of the crossfire menu always works. Just not from a fresh power-up. I contacted TBS support and they could not assist.

2) When binding the Crossfire TX to a Receiver, the option comes up to update the RX. Pressing enter continues the process but there is not update screen showing the progress. I have to keep hitting RTN until the eventually the update process is done and I am returned to Crossfire menus.

Jeff

kilrah, 2017-12-29:
You might need to change the "Max baudrate" setting in the radio setup Hardware page.

closed 2017-12-29 by kilrah

jcutcher01, 2017-12-29:
I am running 115200 the reduced baud rate. Everything works great both flight control and Telemetry but just at power up I have to move the stickers around to get the radio to connect to the module.

jcutcher01, 2017-12-29:
It has to be something with OpenTX and the Lua Script. Can you reopen and confirm?

jcutcher01, 2017-12-29:
It seems changing back the MAX BAUDS to 400000 from 115200 solved the issue and the telemetry is working. I am running TBS Crossfire Beta 2.12 so I assume that might have fixed the original baud rate problem.

schwabe, 2017-12-29:
@jcutcher01 check telem rx errors in the statistics debug screen to see if 400k baud really works

jcutcher01, 2017-12-29:
Running max bauds 400000 on my Horus X10 shows 0 TLM RX ERRORS in the stats screen. Before Crossfire would not run at 400 and I had to drop it to 115200. I wonder if Beta 2.12 solves this somehow. I will monitor this.