opentx/opentx #5582: FM Sensor returns 0

tgreer, 2017-12-31:
I've got some Lua scripts running on 2.2.1 (cca71808) on a X9D.
I have inav running on an omnibus F4 Pro, and CRSF setup both ends.

My FM sensor returns 0 even though it has a value which can be seen on the Telemetry page. 

I've tried calling it by name and by ID
```
modes = getValue("FM")
lcd.drawText(textualRow, y, modes, SMLSIZE)
```
Even calling it with the following gets me blank
```
modes = getValue("258")
lcd.drawText(textualRow, y, modes, SMLSIZE)
```
I got the ID 258 using the following code, which also confirms the right name:
```
       y = y + 8
	local fieldinfo = getFieldInfo('FM')
	lcd.drawText(textualRow, y, fieldinfo['id'], SMLSIZE)
	y = y + 8
	lcd.drawText(textualRow, y, fieldinfo['name'], SMLSIZE)
```

Am I missing something?

r41065, 2018-02-21:
Im Using This Code to get Flightmode Name and Number from Taranis Modes:
searching for this function or is your FM Value a Telemetry Value you get from your FC?

local flm,FM = getFlightMode()	

lcd.drawText(150,10,"FM",SMLSIZE) 	
lcd.drawNumber(lcd.getLastPos()+0,10,flm,SMLSIZE)
lcd.drawText(lcd.getLastPos()+0,10, ":", SMLSIZE) 
lcd.drawText(lcd.getLastPos()+1,10, FM, SMLSIZE)

3djc, 2018-02-21:
Really hard to debug without having your equipment. The getvalue() function does work, there is no question about that, so either there is an issue with CRSF that would put some kind of invalid data that cannot be read correctly by getValue(), or the value is 0 at the very moment it is read (that's two things I can think of, there are surely others, but anyway, all need to be local to your equipement to try to debug)

tgreer, 2018-02-24:
I'll retest it this evening, now I'm back in front of everything.

tgreer, 2018-02-24:
modes         =  getValue("FM")
Sensor named FM under Sensors shows ACRO when called via the above shows 0.

Have renamed the sensor to FMOD and used:
modes         =  getValue("FMOD")
Still shows 0.

tgreer, 2018-02-24:
I think the issue may be that getValue is expecting a number and is being supplied text?

PR #5724 (2018-02-25): added case for UNIT_TEXT used by crossfire FM Sensor
Fixes #5582 
This fixes an issue with Lua scripts being unable to read the FM sensor generated by crossfire which is set to text. Thanks to @schwabe for the fixes, I'm just doing some admin. 

Tested with Taranis X9D, Crossfire Micro TX, Nano RX and inav supplying the telemetry.
closed 2018-02-25 by schwabe