opentx/opentx #5584: X9E+ firmware crashes after startup when built with arm-none-eabi-gcc  7.1

hsteinhaus, 2018-01-02:
I just tried to build OpenTX (branch 2.2) on Fedora 27. F27 brings a quite fresh arm-none-eabi-gcc by default (version 7.1). The build for Taranis X9E+ succeeds and the resulting firmware boots up fine till the throttle/switch warnings. But once these stop conditions have been resolved, the TX immediately reboots again. With switches and throttle now in the correct positions, this results in a rapid reboot loop of about one restart per second. The loop can be stopped by switching of the TX (EEPROM save screen becomes visible for a split second).

After downgrading arm-none-eabi-gcc to the pre-built binaries of gcc-arm-none-eabi-4_9-2015q3 from launchpad, the problem is resolved. A fresh rebuild from the exact same source works fine after a "make flash".

3djc, 2018-01-02:
This is what production firmware should be built with gcc-arm-none-eabi-4_7-2013q3-20130916

hsteinhaus, 2018-01-02:
Unfortunately this information is not present in the "OpenTX 2.2 Linux Build Instructions" in the wiki. This document recommends to install arm-none-eabi-gcc by DNF package manager, which results in gcc 7.1 getting installed.

hsteinhaus, 2018-01-02:
@3djc May I kindly ask to repoen this issue - at least the documentation should get a fix.

3djc, 2018-01-02:
done

closed 2018-01-02 by kilrah