opentx/opentx #5587: Flight modes in companion do not support S1 or S2 six position pot mod for Taranis X9D

kalikraven, 2018-01-02:
It would be nice if companion supported flight mode changes by use of a six position switch in S1, S2 or in the 9+ S3 positions. If there is a way to do this then I am blind and apologize!

Thank you.

kilrah, 2018-01-02:
It does as long as your 6pos is configured correctly. 

[image]

closed 2018-01-02 by kilrah

kalikraven, 2018-01-03:
That will work on the X9D Plus but what about the regular X9 non plus version?

mpaperno, 2018-01-03:
Sure, just make sure it is set up properly in Radio Settings.

[image]