opentx/opentx #5590: S.Bus on PPM pin is SERIAL_8N2 while it should be SERIAL_8E2

DzikuVx, 2018-01-03:
### Should be:
According to specification, S.Bus output should be **SERIAL_8E2**. 

## Is:
In OpenTX 2.2.1, Taranis X9D+, S.Bus output in transmitter bay on PPM pin is **SERIAL_8N2**.

schwabe, 2018-01-03:
Is that something you tested? Because I am pretty sure that it is 8E2. Also since multi and sbus share the `sbusSendByte` function that sends it and multi expects 8E2 too, I am pretty sure it is really 8E2

DzikuVx, 2018-01-04:
Yes, I tested. Did not connected to logic analyzer tho. But was able to correctly read S.Bus packets only with `Serial1.begin(100000, SERIAL_8N2);` on Arduino.

On electrical level:
Taranis -> S.Bus normal (inverted) -> single MOSFET inverter -> ATmega32u4 Serial1 RX

Perhaps I'm missing something, but this is what I'm observing

kilrah, 2018-01-04:
Logic analyzer confirms 8E2.

[image]
[image]

closed 2018-01-04 by kilrah

schwabe, 2018-01-04:
@DzikuVx try 8E1, that should also work. the 10 bit options are often broken. Also you can test without the inverter if you use non-inverterted output from OpenTX.