opentx/opentx #5591: Windows defender claims companion-windows-2.2.1.exe contains a trojan, False Positive?

virtual812, 2018-01-04:
On downloading the latest companion installer from...
http://www.open-tx.org/2017/12/16/opentx-2.2.1
... I get ambushed by windows defender claiming it contains a trojan.

Could possibly be a false positive.
Have tested with other scanners and uploaded it to even more and everything else gives it the all clear.

MD5 of the file as it exists on my disk is...
eec15ef17c708a17398b2c516daead8a

More info in the windows defender popup links to....
https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?Name=Trojan:Win32/Peals.F!cl

I think there is a process for reporting false positives? not sure.

Figured this might be worth posting as there will be a few ppl running the stock Windows Defender.

mpaperno, 2018-01-04:
Hi, can you give us Windows version and date of last Defender definitions update?

Do you have any other AV scanner running in real-time, besides Defender?

Thanks!

mpaperno, 2018-01-04:
Also could you check if the newest nightly build has the same problem?

http://downloads-22.open-tx.org/nightlies/companion/windows/

3djc, 2018-01-04:
Virustotal reposrt full clean 
https://www.virustotal.com/fr/file/22e3dfb785ceb2aa97ec4c318e5a8c5c56795dc7f35f7404cbf48b07a94d1e5f/analysis/

What is worrying for you is that your file doesn't have a proper MD5, it should be :  5f419c09c7f077f37358e00814a6a327 so something on your PC or download path has been tampering with the file

mpaperno, 2018-01-04:
I also get MD5 = `EEC15EF17C708A17398B2C516DAEAD8A` on that download (now and also on a version d-loaded on Dec. 18th).

3djc, 2018-01-04:
Ah yes indeed, unsure why the program I used got it wrong ;s. Retried with M$ utility and I get the same as both of you. Sorry for that. Still not sure why he does have a warning. I'm using data files 03/01/2018

kilrah, 2018-01-04:
Same md5 and up to date windows defender happy about it.

closed 2018-02-20 by mpaperno