opentx/opentx #5598: [Feature request] Open Fileviewer on .lua extension

Gewe99, 2018-01-05:
While browsing the SD Card it is possible to view any .TXT file on a Horus radio.
It would be great if it is possible to open the viewer on .LUA Files

3djc, 2018-01-05:
What is the intended purpose ? Viewing code on such a small screen doesn’t seem very useful

Gewe99, 2018-01-05:
I have several short mixer scripts where I sometimes  change some values .
So I could look easy without a PC which values are  actually in this script...
As the textviever is already implemented it should hopefully not be too much work to implement this feature....

kilrah, 2018-01-05:
The text viewer is limited to a length that would be too short for a script.

RealTadango, 2018-01-08:
You could add a text file with values... but then it is double work.

raphaelcoeffic, 2018-04-12:
Read that text file from your mixer script ;-) or even better, add a display function so your script can be executed standalone to display and possibly change these value.

lshems, 2018-06-27:
Create a text file editor in Lua. I did for the notes.

closed 2019-11-15 by stale[bot]