opentx/opentx #5599: [Feature request] Make  Horus X12 GPS position available via getValue () function

Gewe99, 2018-01-05:
As discusssed in the rocket forum 2 weeks ago, the Horus X12S GPS radio position is only available  when a telelemety link to the modelgps exists. 
An idea from an developer was to handle the radios gps position like a timer...
I am writing this to keep this issue tracked...

Would it be possible to make the inclination information of the radio accessible as well?
Best solution would be a new value like Joystick Jx Jy

Magic-Paul49, 2018-01-07:
Yes, that would be great!

3djc, 2018-01-07:
So great that this has been the case probably since the beginning of opentx horus

Gewe99, 2018-01-07:
Super!!!!!
What is the value widget? Where can I find that?

3djc, 2018-01-07:
Erf, the widget that allow you do display all values !

[image]

Gewe99, 2018-01-07:
[image]

After carrying the radio into the garden it works....
It's nice that the nr. of sats are displayed until  the gps has a fix.
Surprising what is already implemented...

PR #5610 (2018-01-08): X12S : give internal GPS access to LUA
This introduce a function to get intenal GPS data

This can be tested using this (very crappy) widget

[main.zip](https://github.com/opentx/opentx/files/1611526/main.zip)

This fixes #5599
closed 2018-01-08 by kilrah