opentx/opentx #5602: Misbehaviour of GV's increment setting - X9D+, OpenTx 2.2.1

pindeslandes, 2018-01-07:
I set two Adjust GV special functions:
[image]

The GV limits are initially those (unmodified):
[image]

It’s working fine: GV1 increases by 10 or -10 when pressing trim tab right or left.

Then I change the minimum value of GV1:
[image]

And going back to the SF tab, I see that:
[image]

The increments values have been modified but SF1 and SF2 are still working as intended: GV1 increase by 10 or -10 when pressing trim tab right or left.

elecpower, 2018-01-08:
Same issue on radio

closed 2019-06-01 by bsongis

Miami-Mike, 2019-06-14:
I'm not sure if you're aware of this problem and are working on it, or under the impression that it's been fixed, so I'll post it to this thread rather than open a new one:

In Companion 2.2.3, set a GV to Min = -1 and Max = +10.

Set a special function to increment that GV by +1 with RudTrim Right, and another special function to increment (actually decrement) it by -1 with RudTrim Left (or any other choice of switches).

That works. You can use the rudder trim to adjust the GV value up and down within the range of -1 to +10.

But suppose you only want to allow GV values from +1 to +10. Change the Min setting from -1 to +1 so that the GV limits are now Min = 1 and Max = 10. You should now be able to change the value up and down within the range of +1 to +10.

Now go back to the Special Functions page and you'll find that the function that you had set to -1 is now changed to +1 and can't be set below that value. You can no longer reduce the value of the GV using the Adjust GVx > Increment special function.

PR #5609 (2019-06-01): Fixes 5602 Special Functions Inc/Dec Gvar
Fixes #5602 
I made the decision to limit inc/dec to +/-1024 (+/-102.4) to avoid possible unexpected consequences.
In theory if a gvar has range of -1024 to +1024 the max inc/dec is +/-2048.
Though relative to current value even a +/-1024 inc/dec could push the calculated value outside the gvars min/max range. This should be handled by the radio mixer.
commit 882e9638 (2019-06-01): Fixes 5602 Special Functions Inc/Dec Gvar (#6442)

Fixes #5602 (2.3)