opentx/opentx #5604: [Feature request] Adjustable RSSI Warning/Critical Filtertime

Gewe99, 2018-01-07:
The RSSI warning/critical level is already adjustable in the telemety setup. But the warning is very sensitiv. Even being some ms below the limit plays the warning. That makes many pilots nervous, because it often happens only being 300m away. So an adjustable Filter time would be perfect. 
I know that it possible to do that with logic switches,  but I use a lot of them for some planes.
So I would like so use them for other purposes...

schwabe, 2018-01-07:
There is already another issue for that.Search the issues for that. But the general consensus is that this is a bad idea.

closed 2018-01-07 by schwabe

3djc, 2018-01-07:
The issue refrenced is #5223

vitas, 2019-01-30:
Why that is a bad idea? I have many big sailplane models and those low RSSI warnings are so annoying: there are only short spikes in low RSSI signal as in the average everything is just fine. I use two receivers in one plane and still getting warnings. Sounds like rssi filtering is not good or absent..Any rssi anomality should be filtered.

schwabe, 2019-01-30:
@vitas See this post https://github.com/opentx/opentx/issues/5223#issuecomment-330564675

vitas, 2019-01-30:
> @vitas <https://github.com/vitas> See this post #5223 (comment)

A lot of user got those peaks..I get them on any new X frsky receiver, even
on 200 meter, also i never lost a link being at 1 km away. But warning
setting is useless for me, it is annoying like a hell. Almost every one
just set those values very low or switch warning completely off. I suspect
filtering , averaging of RSS frame is not done correctly or does not exist
in opentx

lshems, 2019-01-30:
https://www.justfly.solutions/index.php/full-repository?view=product&id=94:filtered-rssi

filtering RSSI. 

the entire failsafe after 1 second story and telemetry that is referenced is not relevant to wanting this. the telemetry can be gone for seconds without losing control. It is even the normal behaviour. So yes, filtering can be useful.

vitas, 2019-01-30:
thank you! just guess why it could not be part of opentx...