opentx/opentx #5605: Telemetry over Bluetooth on "S-Class" Transmitters

Target0815, 2018-01-07:
What is the output format when the Bluetooth module is set to telemetry on X7S/X10S/X12S? Is this a S. Port mirror only or can Mavlink data be output there?

Target0815, 2018-01-08:
Thanks @kilrah.

Is there still the possibility to issue Mavlink? It would be great if you could feed a ground station directly with simple FrSky components without flight control...

geofrancis, 2018-01-08:
is there any app that can actually use the bluetooth adapter?

kilrah, 2018-01-08:
> Is there still the possibility to issue Mavlink?

OpenTX does not support mavlink in any way currently, and even if it did you'd have to figure out a way to get it from the aircraft to the radio in the first place, which it seems everybody ahs been avoiding so far. 
You can follow #2354 .

> is there any app that can actually use the bluetooth adapter?

FrSky's "Free Link" app on iOS is the only one I know and it's nothing to rave about. Yep for what he wants the work on OpenTX is the least of his worries, the bigger thing would probably for some app he'd like to use to add BTLE support.

Target0815, 2018-01-09:
> OpenTX does not support mavlink in any way currently

Yes, but I have seen several Mavlink statements in the code / header, also in the direction of the Bluetooth interface. That's why I'm asking...

My idea would be to export the telemetry information, which is already available via Smartport on the transmitter (in the telemetry screen), via Bluetooth in Mavlink format.

kilrah, 2018-01-09:
>Yes, but I have seen several Mavlink statements in the code / header,

They are relics of old 9x code. None of it is implemented on FrSky radios.

> My idea would be to export the telemetry information, which is already available via Smartport on the transmitter (in the telemetry screen), via Bluetooth in Mavlink format.

We for sure won't output Mavlink format if the original data is not Mavlink already. Makes no sense trying to convert smart port data to Mavlink. Add smart port support for whatever you want to use the data with instead...

schwabe, 2018-01-09:
If we export parsed data via bluetooth it makes more sense to have a custom data structure that mirrors opentx internal representation of the data (precision, unit, value, name, etc.)

Target0815, 2018-01-09:
> Makes no sense trying to convert smart port data to Mavlink.

I want to explain why this would be useful: we fly OLC ModelGliding. There a triangle is flown down once or several times and the route is recorded via GPS. To be in the top position, it is helpful to be able to follow the flight directly on a map. The GPS positions arrive via Smartport. However, common ground station software can usually only read Mavlink. Therefore, it would be good if you could easily export the Smartport data via Mavlink. Also with an extra data structure, as @schwabe suggests.

I could of course also install an extra Mavlink radio link, but somehow it's not logical to transfer already existing data again...

kilrah, 2018-01-09:
Said ground station software is not able to connect to the Bluetooth module in the radios anyway. 

So work would need to be done on that software for that, at which point adding smart port support to it would be trivial in comparison and much less meaningful than doing a complex conversion in the radio.

mpaperno, 2018-01-10:
IMHO working around the limitations of a GCS (only understanding MAVLINK protocol) with an Open TX radio is not the right way to go (for many reasons).  Such a bridge could much better be implemented in another piece of hardware or on PC itself (assuming that is easier than just modifying the GCS to begin with).   E.g. QGroundControl can connect to a network stream provided by some S.PORT -> Mavlink decoder module.   

I believe the question has been answered so I'm closing the issue for now.   Re-open if it warrants further action.

closed 2018-01-10 by mpaperno

Target0815, 2018-01-10:
> Such a bridge could much better be implemented in another piece of hardware [...]
> S.PORT -> Mavlink decoder [...]

Yes, the view is not wrong, but there is no S. Port => Mavlink decoder, but only in the opposite direction. But whoever has a tip would help me...

kilrah, 2018-01-10:
Nope, someone would have to develop one. Just like someone would have to develop your conversion in OpenTX or an extension to ground station software.