opentx/opentx #5606: Settings/customization not effective immediately at startup (Taranis X9D+)

atomgomba, 2018-01-07:
When I turn on my transmitter I can hear the beginning of the startup sound for a split second at a louder than zero level, even though the global slider I set up for the volume is at minimum. Splash interval is set at ---. This happens on 2.2.1, didn't happen on 2.2.0. If I set the splash interval to 2s then the startup sound plays fully at a non-zero level, because it is shorter than that.

In addition, my custom splash image shows up with the default LCD background color for a split second and then the background color changes to the set one. Splash image shows up regardless of the interval is set to ---. If I set the interval to 2s then the custom splash image shows up with default background color for a split second, then continues to show for the set interval, but with the correct non-default color.

kilrah, 2018-01-07:
Yeah, normal. Splash set to --- is for shortest possible boot time but as a result you get things that happen before the settings are loaded.

closed 2018-01-07 by kilrah

atomgomba, 2018-01-23:
What I've ended up doing is I prepended the startup sound with a 0.3ms silence, so now it begins to play at the volume level set by the slider. Perhaps the firmware itself should postpone playing the sound until it is initialized.

tavdog, 2018-07-16:
I've noticed the startup sound volume thing too, although I don't mind because it indicates what my global volume slider is currently set to.