opentx/opentx #5607: Logical switches are not visible in drop down list

dalarne, 2018-01-07:
Not all logical switches are showing up in the drop down lists for the tab 'Logical Switches'.
This is for Companion 2.2 runing on Windows 10 or 7.
On the Taranis X9E all the logical switches are available.
For instance in the first picture it is not possible to choose the switch L03.
But in the second picture there are a number of switches to choose from.

[image]

[image]

kilrah, 2018-01-07:
Version? A fix for that was made already. If not, close and reopen the window.

dalarne, 2018-01-07:
Version 2.2.1, Dec 17 2017
I tried it on three different computers and downloaded the file twice.
I have reopened Companion multiple times with the same result.
But now, when closing and reopen the window I can choose from switches L01-L05 from any position but the first one.

RealTadango, 2018-01-08:
I think the issue is that you need to re-open the window after you add a LS. Then list do not update when a LS is added / removed.

dalarne, 2018-01-10:
Yes, reopen the window seems to "solve" the issue.
But the drop down lists for the other tabs (mixes, special functions etc.) are updated as expected with/without the new logical switch without the need to reopen the window.
All you have to do is to select a function (logical switches tab), you don´t even have to fill in anything else at that line and the switch is present in the other lists.

mpaperno, 2018-01-10:
This has been fixed early in v2.2.2 with https://github.com/opentx/opentx/commit/76ce186bd4a9e42ee8e336ae7ad601bbda6011c1 .  For 2.2.1 the only workaround is to reopen the window.

Duplicate of #5556.

closed 2018-01-10 by mpaperno