opentx/opentx #5608: X9D+ on screen telemetry data seems to overflow

mpjf01, 2018-01-08:
[MODEL46-2018-01-08.csv.log](https://github.com/opentx/opentx/files/1610595/MODEL46-2018-01-08.csv.log)
[image]

As can be seen the fileds seem to be overfolwed in the bottom row. The three fields are meant to be Alt, Alt+ and Alt- from a second (HP) vario attached to a G-X8R receiver. the top row is altitude from the inbuilt vario. 

The log file is also attached. Logging was only at 5s intervals so it may not be much help. The issue is repeatable, happens after about 15 minutes or so.

3djc, 2018-01-08:
I do not think there is an issue with OpenTx display on that screen that could cause that. In the log attached, there is no abnormal Alta value (maybe because of the 5s log time), but the Vspa value are really  strange, periodicaly, you have either -10.2 or 10.2 changes that do not translate in either Alt or Alta changes, so it looks like your 'a' sensor is somewhat sending randomly bogus data

mpjf01, 2018-01-08:
I redid the test with logging at 0.1s. There's a few examples there of weird altitude values that seemed to have caused the problem.
[MODEL46-2018-01-08.csv.log](https://github.com/opentx/opentx/files/1611051/MODEL46-2018-01-08.csv.log)

mpjf01, 2018-01-08:
It's pretty clear to me that this is a FRSKY sensor issue not an OpenTx problem, so I will take it up with them.

closed 2018-01-08 by mpjf01