opentx/opentx #5614: Bad EEPROM error at boot of 9XR PRO

dadantsou, 2018-01-10:
One week after porting OpenTX 2.2.1 on my 9XR PRO, I encountered a "**Bad EEPROM error**" at boot up. So I tick a button, as suggested, and boom all my models were gone. _In just one sec, everything is gone._ No model, nothing, all my setting was erased. Luckily, I've backed up all my radio settings the day before.

3djc, 2018-01-10:
Yes, when you have this error, you loose everything, since your eeprom which contains all data has corrupted data. It is not the button that make you loose it, it’s already too late when this appears.

The better question is why it got corrupted, but that’s a different story altogether

brandond, 2018-03-06:
I see that the AVR builds have been pulled due to this issue. Is this really all the discussion there is, or is there another issue tracking it? I have a DIY Multiprotocol module on the way, and was hoping to use it with OpenTX on my 9XR - which I understand requires 2.2.0 or better.

powercycle, 2018-03-07:
My understanding is that it has to do with someone willing to maintain the AVR builds.  ER9x works and has multi protocol support.

3djc, 2018-03-07:
It is a combination of several factors:

- for over 2 years now, AVR dev has been virtually non existent in OpenTX. During that period, a lot of major features have been added to arm OpenTX, like virtual inputs for exemple. Current dev is based on those new features, which keep making the gap worse day by day. Basicaly, even if you download a firmware tagged 2.2.1, what you get for avr is the same code as 2.1.x

- the interest for the community, dev as well as users, has moved massively to ARM based radio. We do struggle to find dev ressourses interested in AVR dev (or simply having the required hardware). We also have had virtually no feedback from AVR users during the whole pre 2.2.0 and pre 2.2.1 phases, which did lead to serious bug still present at release because nobody reported them, by lack of user community (it is not a blame by any means, just a fact).

We havent completely finalized our strategy for AVR, so do not take the following for a definitive statement, but our current thinking is to stop AVR support for 2.2, and issue a 2.1 release fixing the existing avr issues.

kilrah, 2018-03-07:
> I see that the AVR builds have been pulled due to this issue.

No it's unrelated.

>  I have a DIY Multiprotocol module on the way, and was hoping to use it with OpenTX on my 9XR - which I understand requires 2.2.0 or better.

2.2.0 supports multimodule on ARM-based radios, but due to what @3djc said of no more development having happened for AVR radios for > 2 years (2.1 didn't even bring anything on top of 2.0 for them actually) there is no multimodule support for a 9XR anyway.

closed 2018-03-07 by kilrah