opentx/opentx #5616: Please make more internal GPS-Data accessible from LUA

Gewe99, 2018-01-11:
Hello together,

great work your fast implementation of the internal Horus GPS accessible by LUA.
Thanks for that so far...

I want to write a widget which shows me the way to my crashed plane....

I tried to calculate the COG CourseOverGround from 'lat' and 'lon'. Dependent on their change over time.
But due to the refreshrate the result is very poor.
I had a look into the radiosource code. The values altitude, speed and groundcourse are already available from the internal GPS.

Here the available structure:
struct gpsdata_t
{
  int32_t longitude;              // degrees * 1.000.000
  int32_t latitude;               // degrees * 1.000.000
  uint8_t fix;
  uint8_t numSat;
  uint32_t packetCount;
  uint32_t errorCount;
  uint16_t altitude;              // altitude in 0.1m
  uint16_t speed;                 // speed in 0.1m/s
  uint16_t groundCourse;          // degrees * 10
};

extern gpsdata_t gpsData;

As I am not a softwarespecialist this is my idea how to implement it.
Perhaps it saves you a few seconds of time...

(Compleatly untested because I am not able to compile...)

I would be very happy if you could implement these 3 more values.

Regards Dirk

LINE 652 api_general.cpp
/*luadoc
@function getTxGPS()

Return the internal GPS position or nil if no valid hardware found

@retval table representing the current radio position
 * `lat` (number) internal GPS latitude, positive is North
 * `lon` (number) internal GPS longitude, positive is East
 * 'numsat' (number) current number of sats locked in by the GPS sensor
 * 'fix' (boolean) fix status
 * `alt` (number) internal GPS altitude in 0.1m
 * `sog` (number) internal GPS speed in 0.1m/s   (SpeedOverGround)
 * `cog` (number) internal GPS Course in degrees * 10  (CourseOverGround)
 
@status current Introduced in 2.2.2
*/
static int luaGetTxGPS(lua_State * L)
{
#if defined(INTERNAL_GPS)
  lua_createtable(L, 0, 7);                                 // DWE 7 correct?
  lua_pushtablenumber(L, "lat", gpsData.latitude * 0.000001);
  lua_pushtablenumber(L, "lon", gpsData.longitude * 0.000001);
  lua_pushtablenumber(L, "numsat", gpsData.numSat);
  if (gpsData.fix)
    lua_pushtableboolean(L, "fix", true);
  else
    lua_pushtableboolean(L, "fix", false);
  lua_pushtablenumber(L, "alt", gpsData.altitude);          // DWE scaling required?
  lua_pushtablenumber(L, "sog", gpsData.speed);             // DWE scaling required?
  lua_pushtablenumber(L, "cog", gpsData.groundCourse);      // DWE scaling required?
#else
    lua_pushnil(L);
#endif
  return 1;
}

3djc, 2018-01-11:
Given there is no compass in x12s, this is done soooo much better by any cheap smartphone, which we all carry....

We may do this change, but that doesn't change the fact that the result on x12 will be 'not great' at the very  least, and order of magniture better on any smartphone

Gewe99, 2018-01-11:
Thank you!!!
From my Garmin GPS I use on my boat, I know that the COG is quiet good, when I walk  faster than 2km/h. So a compass is not really necessary....

Regards Dirk

3djc, 2018-01-11:
Most of those do have compass, however

kilrah, 2018-01-11:
As long as you walk it's OK. Just won't work when stationary.

Gewe99, 2018-01-11:
Good Idea...
I will implement that  in my script...
I always walk fast when my plane crashes 😅

PR #5628 (2018-01-26): Expose more internal GPS data to lua
This fixes #5616
closed 2018-01-26 by bsongis