opentx/opentx #5618: Enhanced trainer mode

LapinFou, 2018-01-12:
Hi Folks,

Since all most everything is possible with OpenTX, I've been happily using a customized trainer mode for many years.
This enhanced trainer mode works like this: hit once the momentary SH button to engage the trainer mode. If I need to take back the control of the plane, I just need to move any stick (except the throttle one) to disengage the trainer mode. This is very comfortable for the instructor.

The programming is explained in details here:
[Trainer Quick Takeback](https://www.rcgroups.com/forums/showpost.php?p=33036059&postcount=9)
and here (French version):
[[Tuto] Ecolage sans maintenir le bouton appuyé](http://frskytaranis.forumactif.org/t1218-tuto-ecolage-sans-maintenir-le-bouton-appuye)

This is perfectly working, but the LS/FS must be duplicated for each model.
It would be awesome if this advanced trainer mode was directly built-in OpenTX.

To summarize:
-  When the trainer mode is engaged, the inputs **Ail**, **Ele**, **Thr** & **Rud** must be replaced with the inputs coming from the student radio according the trainer menu.
- The **trims** of the master radio must be applied on the student's inputs.
- The trainer mode can't be activate if no student signal is present.
- Any movements on **Ail**, **Ele** or **Rud** on the master radio must disengage the trainer mode. An optional parameter to adjust the sensitivity gain would be perfect.
- **Thr** stick movements on the master radio must not disengage the trainer mode. This is important for helicopter training and also when teaching to land an airplane.
- A _trainer status_ indicator (or whatever) in order to get the indication if the trainer mode is engaged or not. This **status** must be usable with any LS. Thus, this status information can be used to play some voice announcements. It can also be use in the **INPUTS** menu in order to use the TRx trainer inputs. This will make possible to use the student sliders or switches. For example with a glider (butterfly), an helicopter (IDLEUP1, 2 or 3D mode), a multi (hover, acro, ... modes).

3djc, 2018-01-12:
Everyone has it it’s own preferred variation of this, not sure of the values of this, given its perfectly achievable using standards otx mechanisms

LapinFou, 2018-01-12:
The big value is this will be easier than duplicating for each model (unless someday we got global logical switches).
Another good point, this will make it accessible to most of OpenTX users.
No every OTX user is a guru with LS. 😄

3djc, 2018-01-12:
Then you can do lua scripts that setup those LS to your liking fairly easily

3djc, 2018-01-12:
Given the (insane) number of practice for trainer (not all train all axis at the same time, some use master radio, other slave,.....),  some kind of wizard is maybe a wiser option

LapinFou, 2018-01-12:
This is a good point. I will have a look.

hmichl, 2018-01-15:
I like the idea of a gain. In a first step it would be cool to be able to choose between a GVAR or fixed weight for trainer function and not only a static weight like currently.

LapinFou, 2018-01-19:
@hmichl 
OK. I will take it in account when creating the LUA Wizard.

closed 2019-11-15 by stale[bot]