opentx/opentx #5621: Feature Request - Horus X10S Backlight off

three6ty, 2018-01-13:
Hi,
Please would it be possible to have the screen backlight go completely off on a timer like you can in FROS. Currently can only set it to dim to around 50% in Open TX. Many thanks.

RealTadango, 2018-01-22:
As far as i understood this is not possible. If you configure the screen off at 0% without a trigger for active.... how do you turn it on again? You could add protections for this though.

voyn238, 2018-02-26:
The stock FrOS firmware by default turned the screen off after X seconds to save power. It would turn back on when you moved the sticks or pressed a key (either, or both - user selectable). After switching to OpenTx I noticed it would no longer do this, only go from bright to not-so-bright. As with any cell phone or tablet, screens are a major drain on the battery. I saw a thread talking about how turning screen brightness below a certain level caused unspecified glitches. I don't want to turn the screen more dim, I want to turn the screen *off*. If FrOS can do it, it seems like it should be doable in OpenTx as well and I would love to see it implemented somehow.

3djc, 2018-02-26:
RealTadango post just above explains why it is not done in OpenTX.  If we force a wake up when you touch stick or buttons, then it kinda defies the purpose, it's gona turn on and off all the time, which probably turns to be more annoying. If we set a key for wakeup, then the script guys complain, if we take a switch, users change them the re quired position might not be available. Not easy to find a perfect design.

Also, at the brightness allowed, the screen drain is minimal

voyn238, 2018-02-26:
On FrOS I had it set to keys-only for wake, because of course if I'm flying I'll be using the sticks constantly, which defeats the purpose. I'm not looking for a specific key to wake it, I'd rather have any key use turn the screen back on. It would be nice to have it maybe fully user programmable, so individuals can tweak it to suit their own needs. Not sure if that's asking the impossible or not. Worst case would probably be someone accidentally turning off the screen without setting a wake condition first...

NevinHK, 2018-03-31:
I agree with this issue. When flying, there’s usually no need to have the screen on at all. With FrOS it timed out and turned off automatically (adjustable in settings). Any key press turned it back on. Mine is chewing through the battery with the screen on all the time.

three6ty, 2018-03-31:
Isn't the screen at something like 45%, not 5% as mentioned above. 
I would love to see this issue resolved in the next update. Be really nice to have the same option as fros please. I can't see why it can't be implemented, but I'm not a programmer so what do I know lol

RealTadango, 2018-03-31:
You can set the off value to 5%.

voyn238, 2018-03-31:
I'm only able to get mine to go down to 45% on 2.2.1, nowhere near 5%. Regardless of what you're able to set the brightness value to, unless the screen is AMOLED, and I don't think it is, the screen's backlight will be going 100% no matter what. The LCD screen just turns darker to make the image less bright. The best way to save battery is to turn the backlight all the way off, which FrOS seems to manage perfectly. 

And if anyone thinks that the screen current draw isn't significant, check the battery usage stats on your phone after a long day. Even with lots of CPU and radio usage (LTE, wifi, bluetooth, etc), the screen uses far more of the battery than anything else. The X10S probably has a bit more current draw on the radio than a phone, but the screen is still drawing a ton of battery use. 

The "why nots" were explained above - why these users or those users would complain based on different "wake up" implementations, but I still think it should be configurable. If the script users don't want it on a key,then don't select that as a wakeup. If other users don't want it on a switch, then disable that too. I just want it to go 100% off when I'm not looking at it (which is most of the time), and turn on when I use the rotary button or the SYS/TELE/MDL/RTN buttons. 

I never used a Taranis, but people say you'd only have to charge them like once a month. I am charging my X10S every few days and it seems a little excessive.

NevinHK, 2018-03-31:
I converted to OpenTx this week and am overall very pleased. Great work guys !

The last few days I have programming directly on the TX, rather than the simulator. My experience is that the Tx battery runs down quite quickly doing this, although I t’s probably drawing more current than when actually flying.

Previously I was using FrOS, and the screen timeout function worked very well, with correspondingly very very long battery life. After converting to OpenTX this is the only issue so far that really baffles me - I had just assumed that a screen timeout would be a given. The screen dim function makes very little difference, mine only reduces brightness by a tiny amount, which leaves the screen drawing a lot of power while I’m flying. FrOS simply uses a function very similar to the OpenTx screen dim settings to wake the screen up. ie touch any key and the screen turns on.

On another note - can someone tell me how to display the voltage of my S6R receiver on the screen ?

219 Seabee Lane
Discovery Bay
Hong Kong.

Tel: +852 9327-8442

3djc, 2018-04-01:
I really wish some of you would do some basic fact checking before posting ... Contrary to what voyn238 is saying, dimming backlight brightness DOES save battery, as we control the cycle time of background leds.

And yes,  I DID do my fact checking, as anyone can quite simply do at home with basic electronic equiment.

X12S, internal module on, external off:
at 100% backlight : 3.1 watts
at 50% baclight: 2.5 watts
at 5% screen : 1.9 watts (that the minimum settings on regular firmware)
at 0% screen : 1.9 watts (done on a custom built firmware)

So on X12S, there is absolutely no battery gain to allow the screen down to absolute dark. I do not have acess to X10, so I wont comment as I'm not able to do actual measurements

three6ty, 2018-04-01:
What you say is correct as far as I can see. However the problem with the x10 is we can only dim the screen to 45%, not 5% like you can on the x12. My dad has the x12 and his screen dims to 5% and is lots dimmer than my x10s.

3djc, 2018-04-01:
Another interesting data showing up in this thread is the FrOS is said to consume less battery, but how factual is that ?

X12S, internal module on, external off:
at 100% backlight : 3.3 watts
at max darknes : 1.9 watt (close to our 5% by the look of it)
Completely off : 1.7 watt

So there is no real technical and measurable 'lower consumption' of FrOS, if user are seing a difference, it is because they use their screen a lot more (widgets, LUA,..) and very likely are much more sound active too).

3djc, 2018-04-01:
So to some up actual FACTS:

- - OpenTx does not drain battery more than FrOS, difference in usage do
- 
- - The screen account for no more than 30% if you have no external module, that figure is  25% if y ou have an external long range module
-
-- On X12S, there will be absolutely no change in battery life when we allow the screen to go completely off
-
-- On X10, you may see at best a 10% gain, assuming you do not use audio and never turn your screen on. Thats very far from the game changing 'days to weeks' that some are hypping

RealTadango, 2018-04-01:
Buy maybe there is a bug in the X10 build that does not allow a value lower then 45%? That is what I read in the comments.

3djc, 2018-04-01:
There is no bug. Due to hardware of X10, value under 45 create flicker, that's why there is that limit on X10

RealTadango, 2018-04-01:
So the real range is 45-100 but it should be displayed as 5-100 ;)... To stop the confusion :)

3djc, 2018-04-01:
No quite, as you cannot get as dark on x10 than you can for x12. What we display is real cycle time. Real values are better :)

closed 2018-04-03 by bsongis

frizzle101101, 2020-07-10:
Can wr further this ides by adding a setting to remove the sticks from the backlight trigger as well as pots and switches...there shld be 3 settings, stick backlight trigger, switch backlight trigger and pot backlight trigger...the hole idea comes from flying at night and moving the sticks turns on my backlight where id like that to only happen really when im scrolling and clicking in menus and such, i just dont like how the sticks turn on the back light when moved but i still want the push of buttons to tuen it on.

3djc, 2020-07-10:
Choose keys

Papa-Echo, 2020-12-08:
> There is no bug. Due to hardware of X10, value under 45 create flicker, that's why there is that limit on X10

But how does FrOS turn the screen off? It is working there.

3djc, 2020-12-08:
The screen can be turned off, what isn't possible are very low values, that work on x12, but create fliker on x10, as your quoted text explains...

PR #5800 (2018-04-03): Rework Horus BL handling
- OFF mode is not available for Horus anymore, as it makes no sense for a radio that requires BL
- if OFF brightness is set to the very mimium, then BL is turned off during off periods

This fixes #5621 and fixes #5403