opentx/opentx #5625: Cant download Companion

troutbum24, 2018-01-14:
I continue to get the following error on my two Windows 10 computers when trying to download the newest version of companion.
This site can’t be reached
downloads.open-tx.org took too long to respond.

3djc, 2018-01-14:
You need to supply more details. downloads.open-tx.org has been up since we moved to the new server. Please given us exact versions you are using.

troutbum24, 2018-01-14:
What kind of information do you need?? I am running a new computer, it is a Lenovo Thinkpad running windows 10. I have not yet installed any Open TX Companion versions on it yet.  I was trying to install companion 2.2 (or any version for that matter) and continue to get the error code (The site cant be reached).  My Taranis Plus is currently running version 2.1.6.

3djc, 2018-01-14:
Not sure what link you are following.

The current 2.2.1 version is here : http://www.open-tx.org/2017/12/16/opentx-2.2.1

The 2.2 version you mention is here : http://www.open-tx.org/2017/05/30/opentx-2.2.0

troutbum24, 2018-01-14:
Hmmm,  I'm not sure why it wont download,  I have been using correct links and tried your link as well. I turned off pop up blockers nothing will work on either of my computers.  My internet connection is strong.  Using google chrome. I even tried internet explorer..

troutbum24, 2018-01-14:
internet explorer gives the following error. (This might be because the site uses outdated or unsafe TLS security settings. If this keeps happening, try contacting the website’s owner.

Your TLS security settings aren’t set to the defaults, which could also be causing this error.)  Not sure what TLS settings I need to use . TLS1.0 is selected There is also TLS2.0 and TLS 3.0

schwabe, 2018-01-14:
TLS 2.0 and TLS 3.0 don't even exist. That is probably SSL 2 and SSL 3. The download links also don't use https (but should, another discussion)

3djc, 2018-01-14:
Both http and https are supported

Certifacte checks ok
https://www.sslshopper.com/ssl-checker.html#hostname=https://downloads.open-tx.org

closed 2018-01-15 by 3djc