opentx/opentx #5629: Redundancy Bus

drummaker, 2018-01-15:
2 things
Why does the transmitter automatically giving me voice announcements when redundancy bus is plugged in and providing Telemetry info

Why is the Telemetry info NOT giving me values for current draw and MAH  used?

3djc, 2018-01-15:
- Because if you use RB, it means safety is of high priority to you, so very much like RSSI alarms are part of OpenTx, basic RB alarm is too

- Probably because you use RB-16 which does NOT include current telemetry

drummaker, 2018-01-15:
I am using RB 10

Yet Maybe I would like change the recordings... Their seems to be no documentation on this.. or am I wrong?

drummaker, 2018-01-15:
under the heading of Values.   It says Over-L - indication of servo status good or overloaded and then below It shows nothing to indicate what value that means...ie Rx2F ?
can you shine any light on this for me?

[image]

3djc, 2018-01-15:
This is not a place for end user support, this is just for Otx bug tracking. Best place for end users support are forums like rcgroup for example. Also this is not quite OpenTx , but more FrSky things, I believe they have a forum too

closed 2018-01-15 by 3djc