opentx/opentx #5630: Companion 2.2.1 Logical Switches V1 V2 Menu Missing Lxx Values

RCdiy, 2018-01-15:
For comparative functions such as a=x,a <b 
L01 to L64 show up in the menu.

For AND, OR, XOR Edge, and Sticky
L01 to L64 does not show up.
Only logical switch configured in a previous editing session show up.

To duplicate this do the following
- New document
- New model
- Edit model
- Logical Switch Tab
- Configure L01 to L07 logical switches in any way 
- L07 try and configure AND with V1 and V2 (you will notice the problem)
- CLOSE the edit model window
- Edit same model again
- Logical Switch Tab
- L07 try and configure AND with V1 and V2 (you will notice that now the configured LS show up)
[image]

RCdiy, 2018-01-15:
[image]

mpaperno, 2018-01-15:
This has been fixed early in v2.2.2 with 76ce186 . For 2.2.1 the only workaround is to reopen the window.

Duplicate of #5556 and #5607

We should make this a sticky or something... :) or maybe leave it open (and tagged somehow as fixed) if people don't search closed issues?

closed 2018-01-15 by mpaperno

3djc, 2018-01-16:
Not sure what the magic receipe here, I keep closing issue for things in the release notes !

RCdiy, 2018-01-16:
I apologize. This was posted in a forum. Kilrah suggested an issue. I made one. Usually I check on rocket chat before.