opentx/opentx #5631: Taranis becomes mass storage after 2.2.1 update

NTTDNV, 2018-01-16:
After updating my Taranis X9D+ to 2.2.1 my radio will only function as mass storage when connected to my computer.  Was working fine before the upgrade. The machine would recognize it and Liftoff would work with the transmitter as input.
Also there is no "masstorage" option to deselect in the radio profile:
[image]

3djc, 2018-01-16:
As mentionned in the release notes : 
"FrSky Radio: massstorage option has been removed. When plugin the radio to USB, you have a choice between accessing SDcard or using the radio as a joystick (choice can also be setup in radio settings)"

closed 2018-01-16 by 3djc

giddy92, 2018-01-17:
How do i choose joystick in through the radiosettings?

3djc, 2018-01-17:
[image]

giddy92, 2018-01-17:
Yea i found it, thanks for the help anyway :)

NTTDNV, 2018-01-18:
Hello

I wasn't getting a choice upon plugging in my TX, it would just go straight
to mass storage mode after my 2.2.1 upgrade.

Turns out I had some issues which have been sorted, now it works as it is
meant to.

Thanks for the reply!