opentx/opentx #5633: Empty/blank tabs in Companion for Debian v 2.2.1

ulfhedlund, 2018-01-16:
The Inputs and Mixes tabs are all empty in my Debian installation of Companion 2.2.1 on Kubuntu 17.10.
Version 2.2.0 works fine but not 2.2.1 or any of the nightly builds for 2.2.2. It makes no difference if I create a new model from scratch or use an existing model file. 

It works on another Debian Linux installation (Linux Mint Ddebian Edition) I have so perhaps it has to do with an incompatible Qt-version or the window manager (as per suggestion from Kilrah).

I have attached a sample otx-file and a summary of my system configuration.

[blanktabs.zip](https://github.com/opentx/opentx/files/1636606/blanktabs.zip)

3djc, 2018-01-17:
First of all you should not be using debian ones, the ones in https://launchpad.net/~opentx-test/+archive/ubuntu/rel/+packages are probably better suited for kubuntu.

That said, even the bionic version does not seem to work properly, we will need to investigate

3djc, 2018-01-17:
I have compiled a version for Kubuntu 17.10 and it is not working either, so fixing this likely requires code changes, that may break other versions. @mpaperno , something to keep in your todo list ?

kilrah, 2018-01-17:
What's the Qt version in that distro?

3djc, 2018-01-17:
cmake -DPCB=X9D ../
CMake Deprecation Warning at CMakeLists.txt:14 (cmake_policy):
  The OLD behavior for policy CMP0023 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- OpenTX 2.2.2 @34e61c45
-- Python found, version: 2.7.14
-- Qt Version: 5.9.1
-- Foxlib found at /usr/lib/x86_64-linux-gnu/libFOX-1.6.so
-- SDL Lib: /usr/lib/x86_64-linux-gnu/libSDL.so;-lpthread; Headers: /usr/include/SDL; Version: 1.2.15

mpaperno, 2018-01-17:
Sounds a lot like a platform issue.  I've only recently started using Qt 5.9.3 on Windows, with no problems so far, but haven't tried the other OSs yet.  Our current "standard" version is 5.7.1 which is what I do most testing with,  5.9.1 is superseded by .3, perhaps there were bugs?  Can you update on that distro?

IMHO this stuff is the reason we should just distribute the Qt libs which we use with our binaries, instead of relying on whatever the system happens to have installed as default.  Chasing down ghost issues with every possible combination is impossible (at least with known Qt version that's one less variable).  Plus one version/package would run on practically any Linux flavor.  :-|

schwabe, 2018-01-17:
Also my local Mac builds use QT 5.9 without issue.

jtaylor2, 2018-01-17:
I have the same problem on my openSUSE 42.3 system using today's self compiled 2.2.  Qt version 5.6.2.

The input and mixer dialogs do not display the line items.  The lines are all populated and the data is there, just not displayed.  If you click on a line it highlights (background changes to blue) and if you double click on the line it brings up the edit dialog with the data displayed.  You can edit the data and it is saved, but is still not displayed on the item line.  If you increase or decrease the font size of a line the blue background changes size, but still nothing is displayed.

I have looked at mixerslistwidget.cpp but that qt is far beyond my ability, but if somebody gives me a hint on where to look I can play with it. 

I have looked at

mpaperno, 2018-01-17:
Could someone post a screenshot?

I wonder if it's just a font/color issue.  Are you guys using custom themes?

You could try running Companion with the alternate Qt style and see if that makes any difference:

`$ companion22 --style=fusion`

Do you see any warnings in the console output when you run Companion from cmd-line, especially when bringing up the model editor for the first time?  

@jtaylor2 Are you using your default platform-installed Qt for building, or a version installed separately?  If the latter, could you also try 5.7.1?

mpaperno, 2018-01-17:
Sorry, that's `-style` with just one dash.  Edited.

kilrah, 2018-01-17:
The issue is that only the first pixel column of each mixer line seems to be drawn, and the problem is with `painter->setClipRect(clipRect);` in https://github.com/opentx/opentx/blob/2.2/companion/src/modeledit/mixerslistwidget.cpp#L214

Commenting it out solves the issue, and doesn't seem to have any detrimental effect (no difference on windows either).
I'm not sure I understand the point fo that ClipRect at all.

jtaylor2, 2018-01-17:
@mpaperno  Yes 5.6.2 is the default for openSUSE.  And running companion with -style=fusion works.

mpaperno, 2018-01-17:
@kilrah Strange, that's the same clip rectangle passed to the document to render into.  I wonder what `textRect ` and `clipRect` actually return.  It's definitely based on application style, as confirmed by @jtaylor2 .  

The clip is meant to ensure nothing gets painted outside the document, though I suppose it could be redundant... I don't remember now if it was there for a specific reason or not (it could have to do with the highlight or drop markers spilling over, perhaps).  I'd definitely test on macOS and some other Linux before just removing it.  I was also testing against Qt 5.2 native on some ancient Linux Mint at the time, so it's possible it had something to do with that.

jtaylor2, 2018-01-17:
```
Same last two lines from mixer dialog

With default style
[D] modeledit/mixerslistwidget.cpp::MixersDelegate::paint():217 - clipRect QRectF(0,0 6x20) textRect QRect(0,196 6x20)
[D] modeledit/mixerslistwidget.cpp::MixersDelegate::paint():217 - clipRect QRectF(0,0 6x20) textRect QRect(0,217 6x20)

With -style=fusion
[D] modeledit/mixerslistwidget.cpp::MixersDelegate::paint():217 - clipRect QRectF(0,0 1246x21) textRect QRect(0,196 1246x21)
[D] modeledit/mixerslistwidget.cpp::MixersDelegate::paint():217 - clipRect QRectF(0,0 1246x21) textRect QRect(0,217 1246x21)
```

mpaperno, 2018-01-17:
Nice thanks.  Obviously a little size difference there!  I guess a question then is why the document isn't clipping properly with `PaintContext.clip = clipRect;` set... which might also explain why the enclosing painter surface itself had to be clipped with `painter->setClipRect(clipRect)`.  I guess one quick workaround would be to "dummy check" the returned size and only apply the clip if it seems "reasonable."  Not a great solution though.  Would be nice to find the root cause.

I wonder which style it is getting in this line, since the sizes are based (partially) on that.

`QStyle * style = (options.widget ? options.widget->style() : QApplication::style());`

jtaylor2, 2018-01-17:
```
For default style
[D] modeledit/mixerslistwidget.cpp::MixersDelegate::paint():207 - QStyle * style = QProxyStyle(0xa24b820)

For style=fusion
[D] modeledit/mixerslistwidget.cpp::MixersDelegate::paint():207 - QStyle * style = QProxyStyle(0x4772560)
```

mpaperno, 2018-01-17:
So the size from this call is wrong,

`QRect textRect = style->subElementRect(QStyle::SE_ItemViewItemText, &options);`

and for some reason is very narrow (though still the correct height).  What happens if you try passing the widget also:

`QRect textRect = style->subElementRect(QStyle::SE_ItemViewItemText, &options, options.widget);`

You could try checking `qDebug() << options.widget->geometry()` and see what that says.  As a workaround we may be able to use the widget's inside width.

The problem could originate on the line above

`style->drawControl(QStyle::CE_ItemViewItem, &options, painter, options.widget);`

If that is not drawing the control properly.  Basically we're kinda faking things here by drawing a dummy list view line and checking how wide a "normal" layout item should be in that view. Then using that size to lay out our actual HTML-formatted document which is then painted into that area instead.  It is (usually) more accurate than trying to estimate the available area in other ways (accounting for scrollbar/padding/etc).

The docs on PaintContext:clip say it is only a "hint" so obviously it is ignoring it and the painter clip rectangle was "properly" limiting the area.

> QRectF PaintContext::clip
> 
> This variable holds a hint to the layout specifying the area around paragraphs, frames or text require painting.
>
> Everything outside of this rectangle does not need to be painted.
>
> Specifying a clip rectangle can speed up drawing of large documents significantly. Note that the clip rectangle is in document coordinates (not in viewport coordinates). It is not a substitute for a clip region set on the painter but merely a hint.

jtaylor2, 2018-01-17:
Passing the widget doesn't help.  Below is with options.widget added:

```
with default
[D] modeledit/mixerslistwidget.cpp::MixersDelegate::paint():210 - options.widget->geometry =  QRect(6,6 1270x222)

with style=fusion
D] modeledit/mixerslistwidget.cpp::MixersDelegate::paint():210 - options.widget->geometry =  QRect(9,9 1262x229)
```

mpaperno, 2018-01-17:
You could try to fake it by using the widget width... I wonder what artifacts that may create though.

```
  QRect textRect = style->subElementRect(QStyle::SE_ItemViewItemText, &options, options.widget);
  textRect.setWidth(options.widget->geometry().width());
```

Try it with and w/out a scrollbar visible in the list view.

jtaylor2, 2018-01-17:
That works  both with and without scrollbar.

Will be off the computer for a while but if you need anything else tested I can do it later tonight or in the morning.

mpaperno, 2018-01-17:
Thanks for testing, Jim.  I guess one workaround might be

```cpp
  QRect textRect = style->subElementRect(QStyle::SE_ItemViewItemText, &options);
  #ifdef Q_OS_LINUX
  // Workaround for buggy Linux native styles (https://github.com/opentx/opentx/issues/5633)
  if (options.widget && textRect.width() < options.widget->geometry().width() * 0.75f)
    textRect.setWidth(options.widget->geometry().width());
  #endif
```

Not sure the `#ifdef` is really needed, it's only a couple CPU cycles and maybe not a bad check in any case.

jtaylor2, 2018-01-18:
Did a quick test on that workaround and it works on openSUSE 42.3.  I'll test it some more tomorrow.

Curious if the problem only occurs on KDE Plasma systems (which use the Breeze style).  I was going to try switching my system to another style, but openSUSE only installs Breeze and I don't want to mess up this system.  Maybe play with my backup system tomorrow.

jtaylor2, 2018-01-18:
I did a little more testing of the workaround and didn't see any problems.  For what it's worth I ran QStyleFactory::keys() on my openSUSE system and it has the following styles available:  Breeze (default), Windows, GTK+, and Fusion.  I tried all of them and Breeze is the only one that returns a bad textRect.

kilrah, 2018-01-18:
Same in my Kubuntu, from the 4 "Widget styles" only Breeze and Oxygen casue the issues, the other 2 are OK. So I guess it's a bug in those styles more than anything.

jtaylor2, 2018-01-30:
@mpaperno are you going to implement a workaround for this as part of something else or should I open a pull request with this one?

mpaperno, 2018-02-06:
Uh, sorry Jim, I let this lag.  I wasn't planning on anything and will be out for a couple more weeks at least anyway, so if you want to do a PR for this, please go ahead.  I'd say keep the Linux #ifdef for now (or can we somehow even better ID the problem platforms?).

jtaylor2, 2018-02-15:
@mpaperno I haven't been able to find a better ID than Linux.  Perhaps style as Breeze and Oxygen are the only ones we know are bad, but then would require a change if/when others turn up bad.

Would be nice to get a fix in 2.2.2 so I've done a PR with your suggested workaround and tested it on openSUSE and Windows 10.   Can always revisit if you come up with something better.

branob, 2018-02-19:
I see the same issue on Linux Mint 18.3 KDE.
companion22 -style=fusion works as workaround

mpaperno, 2018-02-20:
Fixed by https://github.com/opentx/opentx/pull/5695

closed 2018-02-20 by mpaperno