opentx/opentx #5634: Unable to check for updates

eirrick, 2018-01-17:
I am using companion v2.2.1 on ubuntu linux, and I'm getting cannot check for updates error when I try to check for updates or download firmware for my taranis x9d+.

3djc, 2018-01-17:
I cannot reproduce your issue (but I'm trying on debian stretch). Please give as much details as possible

selste, 2018-01-21:
Hi,
same problem here - installed the latest OpenTX version via the provided Ubuntu PPA, and for the last couple of days every time i start the companion software i get the 'cannot check for updates' error dialog.

Enabling debug otput didn't actually help, there are just a couple of messages about the location and registration of shared libraries after startup.

3djc, 2018-01-21:
2.2.1 ?

selste, 2018-01-21:
Yes,
version 2.2.1, via [Packages in “OpenTX Releases”](https://launchpad.net/~opentx-test/+archive/ubuntu/rel/+packages).
Noticed that the firmware downloads weren't working maybe 3 days ago while experimenting with telemetry settings for my Turnigy 9X.

When i reinstalled OpenTX 2.2.1 (must have been around New Year) the downloads worked just fine.

powercycle, 2018-01-21:
I can confirm this as well.

3djc, 2018-01-21:
We are currently working on it

powercycle, 2018-01-21:
If it matters, mine has a date of November 10, 2017 from the about menu.

eirrick, 2018-01-21:
2 days ago, I actually uninstalled the package provided by the ubuntu PPA and installed from the debian package. That fixed the issue. Maybe that helps.

eirrick, 2018-01-21:
I'm running Ubuntu 17.10.

powercycle, 2018-01-21:
For me, it seems I didn't have the actual release version.   Now that I've downloaded that, it shows December 17, 2017 as the date in the about screen.
[image]

powercycle, 2018-01-21:
I got the working version from 
[image]

3djc, 2018-01-21:
It should be ok now

selste, 2018-01-22:
Hi,
sorry - still not working for me.
I'm running the 32 Bit variant, therefore i can't (easily) switch to the Debian package.

3djc, 2018-01-22:
Can you launch it using command line and report what is happening, because it looks ok from here

3djc, 2018-01-22:
Wait wait. What is that November 2.2.1 ? You said it is coming from release ppa, there was no 2.2.1 release in November ????

3djc, 2018-01-22:
I’ll go back in time to November 10 having a look at what was downloaded at that time

selste, 2018-01-22:
Thanks for taking the time to investigate and resolve this problem!

OpenTX is awesome!