opentx/opentx #5635: OpenTx 2.2 - T-Trim still doesn't follow the assigned throttle trim when you cross the trims

mcddd, 2018-01-17:
I'm a fan of crossed trims and would really like to see this problem solved. For electric models the crossed trims work great since the T-Trim function isn't needed. But for gas/glow engines where the throttle trim is only effective at idle, selecting the T-Trim function doesn't follow the selected trim input for the throttle channel. Instead the T-Trim function always affects the THR trim regardless of which trim input device is assigned to the throttle channel. The result is that when cross trims are selected the T-Trim function ends up affecting the elevator channel trim instead of the throttle channel trim.

This has been an open issue since way back on November 29, 2013
https://github.com/opentx/opentx/issues/170

It's been over four years and OpenTx is now on 2.2.1.  Is anybody working on solving this bug?

sande005, 2018-01-18:
Second this request.  Have also opened prior issues, only to have them closed due to lack of interest.

RCdiy, 2018-01-18:
There is support for this in the forums.

RockyBoyRC, 2018-01-18:
Another vote for this one. Love my cross trims, would like to have this behavior consistent between electric and wet fuel models.

Rups63, 2018-01-18:
Another vote to have this feature refined for gas/glow models.

lorenholm, 2018-01-21:
Have used and like them since I built my first Royal Electronics Omega in the early 1980's

Would like it fixed please

robbehymer, 2018-01-27:
I also would like this fixed.

AloftHobbies, 2018-01-29:
Had a few requests for this bug to be patched so thought I would add my 2 cents here. 

Thanks guys!

MarkusPfaff, 2019-03-24:
Is there a reason for having "Throttle Trim Idle Only" as an option in the model setup screen instead of a parameter of an inputs line? It is an option that controls how the trim setting is interpreted for a specific input (the throttle of an IC powered model is the common input where it is needed of course). 

It could be an option of its own (very general solution), but I think it would be more appropriate to add a value to the parameter list "Include Trim" setting for input lines (No, Yes, TrimR, TrimE, TrimA, TrimT, Trim5, Trim6 now, proposal: add TrimThrottleIdleOnly, shortened to TrTio for example).

kilrah, 2019-03-24:
Can't be a choice in the list as that would restrict to one trim control, should be a checkbox.

MarkusPfaff, 2019-03-24:
I realize that I assumed that cross trim would be a radio option then, so that TrimT would always be the trim knob setting throttle trim, be it the knob beside the throttle or the one beside the elevator stick.
But I agree that a checkbox in the dialog for setting up an input line would be the more general solution (and useful for model specific cross trim setup). Maybe it's the more user friendly solution also, because the text besides the checkbox makes the idea more easy to remember. On the other side it's of course more prone to erroneous user input, because there are lots of combinations that don't make much sense. 

By the way, this "Trim Low Side Only" Checkbox could/should grey out or even vanish when the parameter in "Include Trim" is set to NO.

sande005, 2020-03-08:
Keep open. Still needs to be fixed!

mcddd, 2020-03-09:
Even though this has been an unresolved issue that was first reported in November of 2013, it still needs to be corrected.  Please don't ignore it and push it back down the line again.

dgatf, 2020-05-29:
Adding a model option (eg. Throttle trim switch) to select the trim switch to which apply the throttle trim solves the issue (same applies to throttle reverse). This makes no change to current behavior if using default trims

Any reason/disadvantage not to do this?

MarkusPfaff, 2020-05-29:
dgatf, could you please elaborate more on your idea? I do not understand it yet.

dgatf, 2020-05-29:
Since there is an option for throttle trim and trims can be assigned to any input you need to tell opentx which trim you want to act in a different manner

Lets see an example. If you cross trims between elevator and throttle, you do it in inputs but you also need, and this is the new part, to select the trim switch which you want to act as a throttle trim (elevator in this example). opentx can't guess which one is it from the inputs. If you are using throttle trim but not crossing trims you don't need this second step (as it defaults to throttle trim switch). This option would be just below _throttle trim_ in the model menu

A difference from current behaviour is that if you have the throttle trim switch assigned to another input it would act also in that way

dgatf, 2020-05-29:
Well... idle throttle trim with cross trims can be done with the mixes

Why you don't use the mixes instead?

mcddd, 2020-05-29:
Because a checkbox solution is much simpler.  Besides, the TIO checkbox is already there, and it works correctly with standard trims.  After 6 1/2 years of dealing with bandaid solutions, it's way past time to correct the TIO issue when the trims are crossed.

dgatf, 2020-05-29:
Ok. Do you agree with the solution I proposed, the additional option _Throttle Trim Switch_?

mcddd, 2020-05-30:
I don't see how your solution is really much different than the checkbox for the TIO option.  So instead of the TIO checkbox, you are proposing another selection option called Throttle Trim Switch?  If the TIO checkbox doesn't work correctly with the trims crossed, how does your solution solve the problem?

dgatf, 2020-05-30:
> So instead of the TIO checkbox

No, I'm proposing to add an option, not replace. You enable the Throttle trim in a check box, as is now, but also you need select the trim switch you want the throttle trim applied to

mcddd, 2020-05-30:
The specific trim switch for a specific channel can already be selected from the INPUTS screen.  For crossed trims, the "Thr" switch is selected as the trim for the elevator channel, and the "Ele" switch is selected as the trim for the throttle channel.  This works perfectly until the Trim Idle Only function is activated from the checkbox.  The problem is that the Trim Idle Only function stays with the "Thr" switch instead of transferring to the "Ele" switch as it should.  The TIO function checkbox does not enable the throttle trim, it simply alters the way the throttle trim works.  At full throttle stick, the throttle trim has no effect.  At full idle stick, the throttle trim has the full range of trim adjustment.
Or am I still misunderstanding what you are saying?
[image]
[image]
[image]

dgatf, 2020-05-30:
No, sInce the following situation is feasible, you need to redefine the throttle trim switch:

[image]

Only the default trim switch assigned to the throttle stick is known by opentx. Once you change the default throttle trim switch you need to tell opentx the new one

mcddd, 2020-05-30:
Once the trim switch locations have been changed from the default "ON" selection, they show up correctly on the main INPUTS screen as crossed.  And since the crossed trims work, OpenTx obviously knows they are crossed.  At least this is how it works on version 2.2 that I'm using.

Are you saying that later versions have a bug where the crossed trim switch locations don't even show up on the INPUTS screen?

[image]

dgatf, 2020-05-31:
> And since the crossed trims work

Yes, it works with the default relation between trims and sticks (thr-thr, ele-ele....)

>OpenTx obviously knows they are crossed

No. It doesn't know they are crossed, it just apply throttle trim switch to an input which is called Ele and elevator trim switch to another input called Thr. Opentx does not work in the same manner as other firmware radios does. Other radios has an option to cross trims which is different than assigning trims to inputs. This is the main point that explains why you need an additional option if you want throttle trim to work with other stick than the default

> Are you saying that later versions have a bug where the crossed trim switch locations don't even show up on the INPUTS screen?

No, I didn't. I said:

> No, since the following situation is feasible (input 01 with trim to Thr and input 02 with trim to Ele), you need to redefine the throttle trim switch

I just wanted to show you that you can have 2 Thr inputs assigned to different trims switches and opentx can not guess which one is  for the throttle trim

If this is more clear, see the picture bellow with the trims:

[image]

You need tell opentx which trim switch you want to use as the throttle trim switch, it is not feasible to deduce from the inputs. Opentx gives a lot of freedom for programming the models in relation to other firmwares and sometimes you need to feed more data to constraint it properly

closed 2020-10-04 by bsongis

3djc, 2020-10-04:
Please test with 2.3.10N340, we are very close to releasing 2.3.10 and we need to be sure the fix works

PR #5664 (2018-02-11): Apply THR idle trim when cross triming
This fixes #5635 and fixes #170
PR #7677 (2020-10-04): Throttle trim - Fixes 5635
This PR fixes #5635. Though it is not noted in the linked issue, a similar problem occurs with throttle inversed + non default trims

The PR adds a model option, _throttle trim switch_, that allows to redefine the trim switch linked to the throttle stick. The option defaults to the throttle trim switch, so there is no impact on current models

When using non defaults trims (e.g. cross trims), the new trim linked to the throlle stick can be redefined with this option and trims with different behauviour (inversed and idle) are applied to the desired stick