opentx/opentx #5636: X9E SBus Trainer Mode not working

Fi156, 2018-01-18:
Hello,

I am trying to connect two X9E radios via SBUS Trainer mode.
Both radios and both receivers (X8R) are EU-Versions with FW 2.2.1.
(I never flashed the X8R's...)

Configuration file for both receivers:
[X9E_SBusTrainer_Minimal.zip](https://github.com/opentx/opentx/files/1642706/X9E_SBusTrainer_Minimal.zip)

Definition:
Master = Radio bound to the Model-X8R, this radio has the Student bound X8R connected to the middle 3 pins in the external module compartment. (X9D= Signal at the second pin, X9E=Signal at pin 4)
Student = bound to X8R in external module compartment of the master. NOT set to SBUS/Trainer mode.

When I am using an **X9D as master transmitter, everything works fine.** (So my wiring and configuration should be okay.)
When using one of two X9E transmitter as Master, the student receiver will be recognized by the student radio, but no stick input can be seen on screen of the master radio.

What can I do to narrow down the source of the problem?

Fi156, 2018-01-22:
Does the X9E lack the implementation of SBUS Trainer mode at the moment?

Can someone describe the existing problem and give some hints how to solve that?

kilrah, 2018-01-22:
It's not supposed to, but we've seen another report like yours so something is likely broken.

pointu, 2018-06-22:
Grrrr 3djc (on 15apr.),
Same Issue on my x9e, same testing with a x9d+ as master all fine, but with the x9e nothing on the trainers channels, and no signaling trainer on/off.

dohdohpub, 2019-05-16:
Hello,
so far I have always been able to fly by cable in trainer mode without problems, with X9D + Master and 9XR Pro as Slave. I think, the FW version has been 2.1.9 or older. Then I upgraded the slave to 2.2.3 and from there the trainer mode did not work anymore.
I thought it was up to the software update and so I also updated the Taranis, but it still did not work. Then I connected a FrSky XM receiver to connect the transmitter via SBUS in trainer mode, but that does not work either.
Is this a issue that can be fixed or is it a software bug in the firmware?

Gerberus, 2019-07-24:
Same problem here,
trainer with X9E and X12s not working (tested both as slave and master, otx 2.2.3), no signal in OpenTX Trainer menu.
I am aware of the bug regarding sbus input on the X12s, but at the current state i am unable to use trainer, regardless of which radio is master or slave.
Verified receiver SBUS output with DSO, looks good.

WillyZehnder, 2019-09-26:
Same problem with my X9E as master (OTX 2.2.3) - it took a weekend 'til I supposed a OTX-problem :o(

Eldenroot, 2019-12-20:
Any progress? Quite annoying bug... :(

pointu, 2019-12-20:
Hi,
Seams to be an hardware limitation on x9e after a discussion with @3djc can u confirm that ?

3djc, 2019-12-20:
On horus yes, but x9e I can't see why. Finding test equipment to work on this is a bit more difficult, time even more !

pointu, 2019-12-20:
Yes re-reading [post](http://frskytaranis.forumactif.org/t3021p150-tuto-ecolage-sans-fil-avec-opentx-2-1-version-module-jr), "time" is one of the answers.
and yes we have a workaround at this time : CPPM vs sbus

bsongis, 2019-12-21:
SBUS uses USART6, and on X9E there is a BT module which also uses USART6, so it cannot work.

bsongis, 2019-12-21:
No choice for the 2 pins, they can only use USART6. So we need to ensure BT is OFF to use SBUS trainer master

pointu, 2019-12-21:
and as we have no use of BT in x9e as i know (not able to communicate for training )....

commit e2f4e857 (2019-12-21): Attempt to fix #5636
commit a22c765a (2019-12-21): Attempt to fix #5636
commit 818bdff1 (2019-12-21): Attempt to fix #5636
closed 2019-12-23 by bsongis