opentx/opentx #5639: Bootloader 2.2.1 do not work on X7

500mania, 2018-01-19:
after upgrade the firmware to 2.2.1 on Q X7 the bootloader do not start properly

3djc, 2018-01-19:
You need to put much, much more details, as obviouly thousand and thousands of installations have been done without issues. So be very specific of what you did, and what is happening

500mania, 2018-01-19:
I upgraded my X7 with companion 2.2.1 latest stable version
I upgrade the bootloader via sdcard
than...
when I hold both the horizontal trim and switch on the radio the bootloader menu do not start. the display indicate only the version of the bootloader

500mania, 2018-01-19:
now it does not start anymore!!!!

500mania, 2018-01-19:
option are.... multimode, eu, lua, luac, noheli

3djc, 2018-01-19:
works fine here with those options

500mania, 2018-01-19:
how can I to come back to 2.2.0?

500mania, 2018-01-20:
[SOLVED] reflashing the original firmware downloaded from FrSky with Dfuse Demo and then reflashing the bootloader via SD and then reflashing latest firmware with Companion.

closed 2018-01-20 by 500mania