opentx/opentx #5642: Feature: pc to betaflight

gazeddy, 2018-01-21:
How possible is it to add a link from pc to flight controller? We can use Lua scripts to tweak a lot of things as it stands but some we can (not that we need to most of the time) but we have a two way Comms link (on telemetry enabled radios) which personally I would do anything over a wireless link except fw updates.

RealTadango, 2018-01-22:
So you want to connect the PC to the TX over BT and then use the BF app to update the FC? If the BT link for telemetry is 2 way then it should work.

gazeddy, 2018-01-22:
No my idea is to connect the tx to pc using usb cable. Then use it's radio
link (sbus and smartport) for the configuration link

RealTadango, 2018-01-22:
So you want SPort over USB to the PC (like a Serial port). But then the BF can't connect since the protocol is SPORT and not like the FC interface. SPort over USB would be nice for many reasons.

Using the USB on the TX like connecting to the FC will never allow more then LUA does. It is limited to what BF supports over SPORT.

IB1387, 2019-04-02:
What @gazeddy proposed is possible when you use a full-sized TBS crossfire TX. It has bluetooth and a function called "serial bridge". That way I can connect 2 additional wires from my crossfire nano reciever to a free fc uart on 57600baud (if I recall correctly) and connect betaflight configurator wirelessly to the crossfire tx via bluetooth and in turn to the flightcontroller. 
It's not a solution for your case, but it may be an inspiration on how it could work (or you have the money and the urge for the use case to buy a crossfire system).

closed 2019-12-28 by stale[bot]