opentx/opentx #5643: LUA lcd.drawPixmap() on X7 shows incorrect reulsts for images < 25px large

floaledm, 2018-01-22:
Have any of you tried drawPixmap() on the X7 with images narrower than 25 pixels? I can't seem to get any image of 24 pixels wide or less to draw correctly... No problem on the height (as long as it is 64 pixels or less).
Here's the code, the files I used, and a screenshot of the result. test1.bmp is 25 pixels wide. test2.bmp is the same image cropped by one pixel on the right, so it is 24 pixels wide.
```
lcd.drawPixmap(1, 1, "/test1.bmp")
lcd.drawPixmap(30, 1, "/test2.bmp")
```

[test.zip](https://github.com/opentx/opentx/files/1577519/test.zip)

[image]

3djc, 2018-01-23:
Interestingly,  you test2 bitmap does not display at all on x9

floaledm, 2018-01-23:
Thanks for looking into this. Let me know what I can do to help.
The test bitmaps are black and white, since there's no levels of grey on the X7.

3djc, 2018-04-04:
This is still an issue in many case, so this needs to stay open. I have colsed my PR because it was not solving the issue in all cases, and needs to be reworked to provide 100% proper handling

floaledm, 2018-04-04:
My bad. Just wanted to clean up my part. Should have read more carefully.