opentx/opentx #5644: x10s issue in Companion programing SG and SH switches

Crunch69, 2018-01-22:
x10s issue in Companion with SG and SH switches appear to be reversed.  Programing assigned to SH switch operates on the SG switch
In companion edit radio settings >hardware does not allow you to correct the switch settings ie:the SH drop down does not provide 3 position as a setting.

Are the SG and SH switches physically installed incorrectly on the radio?  ( should the switches be swapped from their current positions)

3djc, 2018-01-23:
I'm not sure I understand clearly. On x10s, SH is the two pos momentatry switch, and SG is a short lever 3 pos switch.

I just tested companion 2.2.1, and it is accurate, as can be seen bellow:
[image]

On the radio itself, SG is next to the right slider (RS) and SH is next to the  radio 'handling bar'

kilrah, 2018-01-23:
As you were answered in the chat the hardware for the SH switch on X10 does not support 3 positions.

Crunch69, 2018-01-23:
Shouldn't the SG be next to the slider according to how the rest of the radios switches are laid out?
All the other switches are laid out left to right 
Top Row SF SE SG SH are we saying the top row is SF SE SH SG?
Bottom Row SA SB SC SD

3djc, 2018-01-23:
Yes it is according to frsky descriptions

closed 2018-01-24 by 3djc