opentx/opentx #5653: Companion 2.2 trying to open a .bin eeprom file directly fails

kilrah, 2018-01-26:
Get "can't extractt RADIO/radio.bin" regardless of selected profile. Also while it fails and no window appears there is a document "stub" left open since it asks to convert or save when changing profile/closing

mpaperno, 2018-02-20:
Need a more specific example I think... Just tried a fresh `2.2` build with a few radio types and .bin files and didn't get any errors.  I did get the same error for a while on `2.2` (with any radio/.bin), but that was before 323b30a14b4619861d0bae2add6c3cd1efe46106.

Especially curious to recreate the leftover "stub" issue -- I addressed a lot of that in 0dbad182e8ec0ab3e067aca57b132e416ef8093c but I guess there are still conditions which can cause the MdiChild to not destruct.  The whole situation is a bit awkward really, might be better to only create the MdiChild once the radio data actually exists.

kilrah, 2018-02-20:
Will have to retry on current 2.2 branch... already forgotten, but all the bins I tried at that point failed indeed.

mpaperno, 2018-06-02:
Doesn't seem like an issue anymore.

raphaelcoeffic, 2019-08-18:
@kilrah that happens to me with horus module files, but nothing is left open or the like. Was it the same for you? I think for horus it is just not supported to open just the model file (or radio file only, it's just the same).

kilrah, 2019-08-18:
It was giving the Horus error when dropping a Taranis bin. Likely not a problem anymore, and tons of stuff was reworked since

closed 2019-08-18 by kilrah