opentx/opentx #5654: QtCreator firmware build error

nikola-v, 2018-01-27:
Hello,

followed tutorial for building OpenTx 2.2 and while trying to build firmware error occured:

> 00:59:34: Running steps for project OpenTX...
> 00:59:34: Starting: "C:\Program Files\CMake\bin\cmake.exe" --build . --target all
> [  2%] Built target firmware_translations
> [  2%] Generating font_03x05.lbm
> Traceback (most recent call last):
>   File "C:/Users/nv/Desktop/OPENTX_DEVELOPMENT/opentx-2.2/radio/util/img2lbm.py", line 11, in <module>
>     image = QtGui.QImage(sys.argv[1])
> AttributeError: 'module' object has no attribute 'QImage'
> radio\src\bitmaps\128x64\CMakeFiles\9x_fonts_1bit.dir\build.make:106: recipe for target 'radio/src/bitmaps/128x64/font_03x05.lbm' failed
> mingw32-make.exe[2]: *** [radio/src/bitmaps/128x64/font_03x05.lbm] Error 1
> CMakeFiles\Makefile2:364: recipe for target 'radio/src/bitmaps/128x64/CMakeFiles/9x_fonts_1bit.dir/all' failed
> mingw32-make.exe[1]: *** [radio/src/bitmaps/128x64/CMakeFiles/9x_fonts_1bit.dir/all] Error 2
> Makefile:150: recipe for target 'all' failed
> mingw32-make.exe: *** [all] Error 2
> 00:59:36: The process "C:\Program Files\CMake\bin\cmake.exe" exited with code 2.
> Error while building/deploying project OpenTX (kit: Desktop Qt 5.8.0 MinGW 32bit2)
> When executing step "CMake Build"
> 00:59:36: Elapsed time: 00:01.

Building with Qt 5.8.0 MinGw 32.
Any ideas anyone?

mpaperno, 2018-01-27:
It looks like you don't have PyQt installed (or it can't be found, anyway).  This step of the build is generating font data files from images using Python and PyQt.

nikola-v, 2018-01-28:
Thank you for helping out!

mpaperno, 2018-01-29:
Good to hear! I'm not sure how much of this is required, but I have 3 `PATH` entries for Python in my build environment:

`PATH=...;E:\Devel\Python2;E:\Devel\Python2\Tools;E:\Devel\Python2\Lib\site-packages\PyQt4;...`

closed 2018-01-29 by mpaperno