opentx/opentx #5655: problem about  bulid code

skyyuzhang, 2018-01-27:
O! hi! thanks for your reading. first of all! I'm a new guy for open tx !!!  and I'm trying to bulid code on my computer whith Qt . by the way my computer system is win10.  but  as I said , I am a new guy and  I think your compiling instruction is so simple that I an not understand it!!!!!  I know it need  about c++ or something like that and I also sure on programing  code is ok .  Just do not know how to compiling it .(a little more complicated). Here is my thought that could  show me a guide which is more specific more detial than the github web's  bulid instruction in text or video ,both of it will be great!!!  To let me know how to bulid it.   I will be very very very greatful!!!     Thank you!!!!!

3djc, 2018-01-27:
Max has gone through great efforts to propose a detailed guide:

https://github.com/opentx/opentx/wiki/OpenTX-2.2-Build-With-QtCreator-%28Win-Linux-OSX%29

As far as I know nothing more detailed exists

mpaperno, 2018-01-27:
Builds fine on Win 10, same steps as Win 7.  Follow the instructions, it should work eventually.  

I would not call it especially simple though, even after setting up many builds it seems like there's always some little niggle to get around. It will be a learning experience.  If you have a specific compilation problem then open a new issue here, but please do your homework first.  It does work, so assume any issue is in your setup.  Instructions can always be improved of course, suggestions welcome.

closed 2018-01-27 by mpaperno