opentx/opentx #5657: Remark column in Logical Switches

dmalcman, 2018-01-28:
I would dearly love to see  a Remarks column (or a label)  added in the Logical Switches, (in companion only), as this would help greatly to understand the functionality of the given Logical Switch.

This would be specially helpful when visiting or reviewing the Logical Switches in weeks or months after configuring the function.

A column of say 9 characters long or even less would be fantastic.

I am an Open Tx addict, could no see myself with any other radio operating system.

Your hard work is very much appreciated.

Daniel

kilrah, 2018-01-28:
With the current storage paradigm it is not possible to store any extra information than what's stored in the radio. 

Will change with the upcoming storage upgrade.

closed 2018-01-28 by kilrah

dmalcman, 2018-01-28:
Thank you for taking my request into cosideration