opentx/opentx #5660: Question: increased telemetry refresh rate

us1m41, 2018-01-29:
I am working on a position sensor that i am trying to get from the model back to my X9D+ using the A2 telemetry port. The refresh rate of the telemetry data seems to be sluggish (estimate at 250-500ms?). 

Is there any way I can make this more responsive? 10hz (or faster)?

Thank you!

Awesome software!

3djc, 2018-01-29:
No, this is governed by the receiver itself, isn't fast has you have been seing, and also depends on the amount of other telemetry data being sent by the receiver.

closed 2018-01-29 by 3djc

kilrah, 2018-01-29:
You need to use the smart port if you want anything faster.

us1m41, 2018-01-31:
thank you for the continued responses. 

It appears that the receiver is both filtering and has a very slow refresh rate. 

-Who can I talk to about receiver hardware and software?  

- does anyone have a recommended small, light, simple s-port voltage sensor they recommend. I have a VFAS40 and it is to big/ heavy.   Would this work: https://alofthobbies.com/analog-voltage-sensor-for-frsky-x8r-receivers.html

-if I switch over to s-port, what should my expectations of refresh rate be?

(I am also reaching out to the vendor of the sensor above). 

Thank you again!

kilrah, 2018-01-31:
Yes it works. Refresh rate is determined by the sensor. There are several arduino-based open source sensor implementations on which you can change anything you want. Including probably using a more appropriate data field than just voltage :)