opentx/opentx #5662: Feature Request: Allow title bar customization on QX7 (128x64 screen size)

wr0ngway, 2018-01-31:
This may be a lack of knowledge on my part or a bug in the current implementation (2.2.1), but I can't figure out a way to add a voltage display (from telemetry VFAS) in the title/top bar on the default model view (i.e. beside Tx batt voltage, signal).

I'm running Opentx 2.2.1 on a QX7S. The Model Telemetry screen doesn’t have an option for Voltage source.  My telemetry is set up and working correctly - I can show VFAS on a custom screen, just can’t find the option to turn it on for the default screen toolbar.  I have discovered sensors within the model telemetry config, and VFAS is showing as a sensor in there. Aside from the sensors themselves, all I see in the model telemetry config screen are options for RSSI alarms and Vario settings. The docs say one can do it (search title bar), but I’m guessing they are out of date or specific to the X9D: https://opentx.gitbooks.io/manual-fo...ain_views.html

3djc, 2018-01-31:
X9 has nearly twice the pixel count of x7, and therefore much more screen space to play with. I'm not sure what you call default screen toolbar,

If you are thinking of this this is quite packed already

[image]

wr0ngway, 2018-01-31:
By default screen toolbar I mean the screen that comes up when you first power on, i.e. the green in this image:
[image]

3djc, 2018-01-31:
That soace is for flight mode name

wr0ngway, 2018-01-31:
Ok, I've never actually seen it populated with anything else except some signal bars, but I only use it for quadcopters and I haven't used it that long so ... :)  Would be nice if one could customize the widgets in there though so one could exclude the data they didn't care about and add the data thats important to them.

3djc, 2018-01-31:
[image]

3djc, 2018-01-31:
There is no customisation at all for this page. It could be added, but would need a lot of work to have it smart enough to not waste space, and alos require an eeprom change

wr0ngway, 2018-01-31:
My goal here is to satisfy my laziness to be able to check quad voltage without having to hit any buttons to get into another screen. :)  I've noticed that the view chosen for the area below (where stick positions are shown) persists between power cycles.  Is it possible to add a screen for telemetry there?

lshems, 2018-01-31:
Could be solved if they would execute the autoexec.lua in the '/scripts/telemetry' directory on startup.

schwabe, 2018-01-31:
@lshems feel free to submit a pull that allows starting with the first telemetry screen on boot.

lshems, 2018-02-01:
@schwabe I would love to, if I could understand all this toolchain, github and c programming stuff.

Spent several days one time trying to set it up. Even made a dedicated Ubuntu laptop for it and installed all just to be able to compile. Never got it to work.

Can't grasp it, can't get it to work. 

Sorry, my contribution is limited to ideas, some good, some bad, some ugly, all in the eye of the beholder.

wr0ngway, 2018-02-01:
I may give it a shot, but I'm completely unfamiliar with the codebase, and the last time I wrote C/C++ was 15yrs ago! :D  I was able to get the toolchain setup and completed a make, but it would be helpful if you could give me some pointers to the files I'd want to edit to do this.

teckel12, 2018-06-03:
Sorry about digging up this old issue.  But allowing the user to auto-select a default Lua script would be a handy feature.  I wouldn't make it a required `autoexec.lua`, but instead have a parameter where Lua script could be selected (as it's selected for custom screens currently).  It could even a setting per model.

I'll give another attempt to setup a dev environment for OpenTX builds.  I had problems in the past that maybe a fresh start will resolve.

3djc, 2018-06-03:
That would require eeprom space to store that choice, so cannot happen before 2.3

closed 2019-11-15 by stale[bot]