opentx/opentx #5668: Feature request: add globalTimer to getGeneralSettings()

Storchy1, 2018-02-05:
Is it possible to add one more Element to the table of the Lua getGeneralSettings().
The Element is g_eeGeneral.globalTimer.
It would be perfect to have an access to g_eeGeneral.globalTimer over Lua scripts to reset the value
A LUA script can then automatically reset the GlobalTimer, if the battary was charged.

3djc, 2018-02-05:
You have access to timers using model.getTimer and model.setTimer, so sure what more you need

3djc, 2018-02-05:
OK, I see what you want to do, sorry

3djc, 2018-02-05:
Addiong it to getGeneralSettings() would allow you to read the value, but you still need a way to reset it, which I do not think LUA allows currently, so you would need a new function for that

Storchy1, 2018-02-05:
Yes, you are right, the first step is to get the value over Lua Script function getGeneralsettings(). I think this will be possible without to many effort. Bit the second step (reset the value) will be a complex process.

Storchy1, 2018-02-06:
Thank you very much

closed 2018-02-11 by kilrah

CapnBry, 2019-03-12:
I realize this is closed due to the pull request being accepted, but the current elapsed time is not a "setting" as you'd expect as part of the return of `getGeneralSettings()`. I'm looking for the Throttle Time and Throttle Percent Time in LUA and we don't have those either. Wouldn't this be better served by a new function `getSystemTimer(which)` or `getGlobalTimer(which)` that can return:
```
0: return g_eeGeneral.globalTimer; break; // global timer
1: return s_timeCumThr; break;  // throttle timer
2: return s_timeCum16ThrP/16; break; //throttle percent timer
```
(obviously it would return lua_pushinteger() for them but you get the idea). `resetGlobalTimer()` can remain the same or allow for resetting of the throttle timers as well with similar code.

Storchy1, 2019-03-13:
Of course, it would be better to have this two functions to get all timer values to LUA. Please ask kilrah for these function. Perhaps he has time to develop it.

CapnBry, 2019-03-13:
Glad you approve! I've gone ahead and made the changes myself (#6369). Tested out working on Taranis Q X7.

PR #5669 (2018-02-11): Add gtimer support to LUA
- allow to read gtimer value using getGeneralSettings()
- allow to reset it using resetGlobalTimer()

This fixes #5668