opentx/opentx #5671: Open tx 2.2.1

piocarlos, 2018-02-05:
Have no start up welcome to open tx , even though the wav file exists on the sd card sounds.

kilrah, 2018-02-05:
Not an OpenTX issue. Wav file changed name between 2.1 and 2.2 by the way.

closed 2018-02-05 by kilrah

piocarlos, 2018-02-05:
ok how do I get the welcome like I used to then please.

piocarlos, 2018-02-05:
Hi, how do I get the open tx welcome back like I used to have in 2.0.
please?

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

3djc, 2018-02-05:
Maybe reading the release notes would be a good first step.

piocarlos, 2018-02-05:
nothing in the release notes about this issue!

kilrah, 2018-02-05:
No?

> MISCELLANEOUS
> [...]
> tada.wav has been renamed to hello.wav

piocarlos, 2018-02-05:
Same question how do I set it to play at switch on ?

piocarlos, 2018-02-05:
Same question how do I make it play that at start up?

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

kilrah, 2018-02-05:
You don't do anything, if it's there it plays. Just like before apart from the name.

piocarlos, 2018-02-05:
Ok thanks buddy

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>