opentx/opentx #5677: SxR LUA - S8R calibration value

Antury1, 2018-02-08:
When calibrating S8R and using the SxR script, displays incorrect values XYZ that do not match the values when connecting the receiver to the PC ( here displays correct ) ). At S6R is not a problem. Also, when using FrSkyOS on the transmitter there is no problem. Have any suggestions for solutions please?
[image]

3djc, 2018-02-08:
According to FrSky site, there is a firmware that "solve the issue about lua script", but I have not tested it myself

Antury1, 2018-02-08:
I tried more S8R with the last available FW ver.171018 and  also  ver.171213  and unfortunately still the same :-(
The problem is on FW as you confirm.

Antury1, 2018-02-08:
can you test it myself and give me the information please?

3djc, 2018-02-08:
Sorry I do not have a colorscreen radio, but I tested latest firmware with latests supplied S8R calib script (170925) and it does not work either.

You need to take that to FrSky I guess

Antury1, 2018-02-08:
Thank you for your quick consultation and recommendations. I'll hand it directly to FrSky. BR

3djc, 2018-02-08:
Please copy the new script file from our repository: https://github.com/opentx/opentx/blob/2.2/radio/sdcard/horus/SxR/SxR_Calibrate.lua

commit afb0ca38 (2018-02-08): Compensate for S8R firmware bug
This fixes #5677