opentx/opentx #5678: Simulator typo X12S

Oldgibber, 2018-02-08:
On the Horus X12S simulator the button which should be labelled "MDL" is actually labelled "MOL". The X10 is correctly worded. (version 2.2.2 nightly)
[image]

schwabe, 2018-02-08:
The image reproduces the real hw. The real x12s has mol on that button.

Oldgibber, 2018-02-08:
Nooo! I have my 12S in front of me and it says MDL.

schwabe, 2018-02-08:
Google images disagrees with you. Might be that FrSky changed that on later models. But what the simulator shows is not wrong.

Oldgibber, 2018-02-08:
Might be right for earlier models, but not so now. I'm trying to update 200 pages of documentation, and typos like this make it difficult both for me and the inexperienced user.

Oldgibber, 2018-02-08:
At least leave this open so that others can discuss.

kilrah, 2018-02-08:
My production Horus says MDL, I think MOL was on the beta ones.

Oldgibber, 2018-02-08:
Replacement rotary encoder. £3.60 in the UK. Having heard of many breaking I bought a spare.

kilrah, 2018-02-11:
We don't have the source files, @mhotar can you have a look?

closed 2019-11-15 by stale[bot]