opentx/opentx #5680: In Companion v2.2.1, when editing logical switches, the last switch created in the drop-down menu is missing.

Ceeb182, 2018-02-09:
This is not a blocking bug.

It's @phil-86 who found it. @phil-86 is a member of the french forum http://frskytaranis.forumactif.org/.

**How to reproduce the bug in Companion v2.2.1 :**  
- Open Companion v2.2.1
- Create a new model
- Open "Editing model" window
- Go to the "Logical Switches" tab
- Write `L01` and `L02` like this :
  `L01   a>x    V1=E1:Gaz      V2=20`
  `L02   a<x    V1=E1:Gaz      V2=80`
- With `L03` I want to do "`AND L01 L02`", but for the `L03` line in the drop-down menu of V1 and V2, there is no `L02`, there is only `L01`.
- You have to close the "Editing model" window and re-open it to see `L01`, `L02` and `L03` in the drop-down menu of V1 and V2 for the `L03` line

3djc, 2018-02-09:
Know issue, has been fixed, the fix will be available in 2.2.2

closed 2018-02-09 by 3djc

Ceeb182, 2018-02-09:
Ok.  Thank you.