opentx/opentx #5682: opentx 9x128 green ed040502-h16d invalid eprom filesystem

s3melle, 2018-02-11:
opentx 9x128 green ed040502-h16d invalid eprom filesystem #5092 

Still have bad eeprom while updating and can't save (read/write) radio parameters.

semelle@dobian:~$ companion22
[D] Translations::installTranslators():101 - Locale name: "fr_FR" language: "français" country: "France"
[D] Translations::tryLoadTranslator():158 - Installed translation file "qt" from "/usr/share/qt5/translations" for "fr_FR"
[D] Translations::tryLoadTranslator():158 - Installed translation file "qtbase" from "/usr/share/qt5/translations" for "fr_FR"
[D] Translations::tryLoadTranslator():158 - Installed translation file "companion" from ":/translations" for "fr_FR"
[D] registerStorageFactory():54 - register storage "bin"
[D] registerStorageFactory():54 - register storage "eepe"
[D] registerStorageFactory():54 - register storage "hex"
[D] registerStorageFactory():54 - register storage "otx"
[D] registerStorageFactory():54 - register storage "sdcard"
[D] SimulatorLoader::registerSimulators():55 - [simulator.interface.loader] Searching for simulators in "." matching pattern ("-simulator.so")
[D] SimulatorLoader::registerSimulators():83 - [simulator.interface.loader] Found libraries: ("none")
[D] SimulatorLoader::registerSimulators():55 - [simulator.interface.loader] Searching for simulators in "/usr/local/lib/companion22" matching pattern ("-simulator.so")
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in "libopentx-9x-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-9x" simulator in "/usr/local/lib/companion22/libopentx-9x-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in "libopentx-9x128-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-9x128" simulator in "/usr/local/lib/companion22/libopentx-9x128-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in "libopentx-9xr-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-9xr" simulator in "/usr/local/lib/companion22/libopentx-9xr-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in "libopentx-9xr128-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-9xr128" simulator in "/usr/local/lib/companion22/libopentx-9xr128-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in "libopentx-9xrpro-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-9xrpro" simulator in "/usr/local/lib/companion22/libopentx-9xrpro-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in "libopentx-ar9x-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-ar9x" simulator in "/usr/local/lib/companion22/libopentx-ar9x-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in "libopentx-gruvin9x-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-gruvin9x" simulator in "/usr/local/lib/companion22/libopentx-gruvin9x-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in "libopentx-mega2560-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-mega2560" simulator in "/usr/local/lib/companion22/libopentx-mega2560-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in "libopentx-sky9x-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-sky9x" simulator in "/usr/local/lib/companion22/libopentx-sky9x-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in "libopentx-x10-simulator.so"
register theme Default
register theme Darkblue
register layout Layout2x2
register layout Layout1x1
register layout Layout2x1
register layout Layout2P1
register layout Layout2x4
register widget Text
register widget Timer
register widget ModelBmp
register widget Outputs
register widget Value
register widget Gauge
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-x10" simulator in "/usr/local/lib/companion22/libopentx-x10-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in "libopentx-x12s-simulator.so"
register theme Default
register theme Darkblue
register layout Layout2x2
register layout Layout1x1
register layout Layout2x1
register layout Layout2P1
register layout Layout2x4
register widget Text
register widget Timer
register widget ModelBmp
register widget Outputs
register widget Value
register widget Gauge
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-x12s" simulator in "/usr/local/lib/companion22/libopentx-x12s-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in "libopentx-x7-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-x7" simulator in "/usr/local/lib/companion22/libopentx-x7-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in "libopentx-x9d+-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-x9d+" simulator in "/usr/local/lib/companion22/libopentx-x9d+-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in "libopentx-x9d-simulator.so"
[... 79 lines cut ...]
[D] OpenTxEepromInterface::load():186 - trying OpenTX for FrSky Horus import... wrong size (9740/0)
[D] OpenTxEepromInterface::load():186 - trying OpenTX for 9XR-PRO import... wrong size (9740/524288)
[D] OpenTxEepromInterface::load():186 - trying OpenTX for M128 / 9X board import... wrong size (9740/4096)
[D] OpenTxEepromInterface::load():186 - trying OpenTX for 9X board import... wrong size (9740/2048)
[D] OpenTxEepromInterface::load():186 - trying OpenTX for 9X board import... wrong size (9740/2048)
[D] OpenTxEepromInterface::load():186 - trying OpenTX for M128 / 9X board import... wrong size (9740/4096)
[D] OpenTxEepromInterface::load():186 - trying OpenTX for ar9x board / 9X import... wrong size (9740/524288)
[D] OpenTxEepromInterface::load():186 - trying OpenTX for Sky9x board / 9X import... wrong size (9740/524288)
[D] OpenTxEepromInterface::load():186 - trying OpenTX for Gruvin9x board / 9X import... wrong size (9740/4096)
[D] OpenTxEepromInterface::load():186 - trying OpenTX for MEGA2560 board import... wrong size (9740/4096)
[D] StorageFormat::setError():109 - [bin] error: Fichier EEPROM invalide /tmp/18723-temp.hex
[D] EepeFormat::load():38 - [eepe] No EEPE header
[D] OpenTxEepromInterface::load():186 - trying OpenTX for FrSky Taranis X9D+ import... wrong file system
[D] OpenTxEepromInterface::load():186 - trying OpenTX for FrSky Taranis X9D import... wrong file system
[D] OpenTxEepromInterface::load():186 - trying OpenTX for FrSky Taranis X9E import... wrong file system
[D] OpenTxEepromInterface::load():186 - trying OpenTX for FrSky Taranis X7 import... wrong file system
[D] OpenTxEepromInterface::load():186 - trying OpenTX for FrSky X10 import... no
[D] OpenTxEepromInterface::load():186 - trying OpenTX for FrSky Horus import... no
[D] OpenTxEepromInterface::load():186 - trying OpenTX for 9XR-PRO import... no
[W] OpenTxEepromInterface::checkVariant():976 - wrong variant ( 0 )
[D] OpenTxEepromInterface::load():186 - trying OpenTX for M128 / 9X board import... version 216 ko
[W] unknown():0 - QXcbConnection: XCB error: 3 (BadWindow), sequence: 14909, resource id: 4195086, major code: 40 (TranslateCoords), minor code: 0
[D] OpenTxEepromInterface::load():186 - trying OpenTX for 9X board import... version 216 variant 0 ok
[W] unknown():0 - QXcbConnection: XCB error: 3 (BadWindow), sequence: 15025, resource id: 4195095, major code: 40 (TranslateCoords), minor code: 0
[D] main():165 - COMPANION EXIT 0

kilrah, 2018-02-11:
#5586

closed 2018-02-11 by kilrah

s3melle, 2018-02-16:
I don't understand.
is there any correction ?

kilrah, 2018-02-16:
No, just that your issue is a duplicate.