opentx/opentx #5691: Trim Center Beep does not work on the thottle trim when Trim Idle Only is selected

mcddd, 2018-02-12:
When the Trim Idle Only function is selected, the center beep doesn't work for the throttle trim, even if the center beep box is checked on the SETUP screen.  This has been a problem since at least 2.0 and is still a problem with 2.2.  It would be nice to get the center beep working correctly.

3djc, 2018-02-13:
That's interesting because we where asked yesterday to also remove it when cross triming. The centre trim beep is purposefully removed, it is not a bug.

What would center trim mean in a throttle trim scenario ?

mcddd, 2018-02-13:
"Allows me to setup my radio so that it does what I want it to do exactly the way I want it to do it: No compromise, no workarounds, no having to think like someone else."

"If I can dream it, I can make it happen!"

So do those first two paragraphs sound familiar?  They should.  These are the top two reasons for using OpenTx as listed on the OpenTx University site on the Introduction page.
http://open-txu.org/home/undergraduate-courses/introduction/why-upgrade/

The center trim beep provides a quick audible indicator of the center trim position, regardless of which trim is being adjusted.  This is especially true while the aircraft is inflight and it's inconvenient to look at the trim displays since glow/gas engines often require further throttle trim adjustments to achieve a slow enough idle speed for landing.  

Since the center throttle trim beep can be turned on or off from the SETUP screen without the Trim Idle Only activated, why wouldn't the same selection box be able to turn the center beep on or off with the Trim Idle Only activated?  Those that don't want the center beep can simply uncheck the box, and those (like me) that do want the center beep can check the box.  Eliminating the choice goes against the top two reasons for using OpenTx that were listed at the beginning of this reply.

3djc, 2018-02-13:
First of all, you probably realize it, but while been a great site, opentx university has nothing to do with us ...

Second and more importantly, I'm very interested in the checkbox you mention that allow you to turn on or off center trim, would you join a screenshot or a photo ?

mcddd, 2018-02-13:
Kilrah, Thank you for explaining how the center beep works!  I had always assumed the center beep check boxes were for the trim functions, not the stick functions.  This completely explains the behavior.

RCdiy, 2018-02-13:
rcdiy 1:46 PM
Cross Trims - works now. Noticed that the trim centre does not work as expected. 
When cross trimmed the trilT does not have a trim centre (remember this has been crossed so actually is ELE trim)... **Now that trimT is ELE trim should it not have trim centre?**
  
Also the TrimE, which is now THR trim has a centre when it probably should not... Tried this on the latest companion nightly.... **Now that TrimE is THR trim should not the centre beep be removed?**

mcddd, 2018-02-14:
RCdiy is correct.  The Trim Idle Only selection removes the faint chirp at trim center from the TrmT button, regardless of whether or not the trims are crossed in the software.

So with crossed trims the active elevator trim (TrmT) does not have a faint center chirp, which it definitely should.

And with crossed trims the active throttle trim (TrmE) retains the faint chirp at center trim.  I would like to see this center chirp remain active, but I can live without it if it's that important to the opentx team to keep it silenced.

Sorry for the confusion earlier but it wasn't until Kilrah pointed out how the center beep worked that I realized there is a difference between a loud center stick beep and a faint center trim chirp.

3djc, 2018-02-14:
Which is exactly what #5693 fixes